noble/noble #273: Notify not working after reconnecting (fault?) device

manuelmitasch, 2015-11-20:
I have an issue that a device enters a (possibly) faulty state after I have not properly disconnected it. After that I'm unable to connect to the device on the first try. After a reconnection the service + characteristics discovery callbacks are triggered twice and the notifications are not working.

Here is my code, console log, hci dump: 
https://gist.github.com/manuelmitasch/c50cf2a8126446306f09

Here is the console log:

```
$ sudo node test-catch.js
poweredOn
dicovered peripheral: 000780790ac9
connected to peripheral: 000780790ac9
DISCONNECTED. RECONNECTING...
connected to peripheral: 000780790ac9
discovered juggglow service
discovered juggglow service
discovered characteristics
discovered characteristics
# Throw ball which should trigger events => no events
# reset device through a usb cable connection
DISCONNECTED. RECONNECTING...
connected to peripheral: 000780790ac9
discovered juggglow service
discovered characteristics
# throw ball => events are working
EVENT: 02
EVENT: 01
```

* Noble version:
  * 1.2.1
* Node version:
  * v0.12.7 (but also tested on v4.2.2)
* Bluetooth device:
  * Juggglow - smart juggling ball
* Bluetooth dongle: 
  * Bus 004 Device 003: ID 0a5c:21e8 Broadcom Corp. BCM20702A0 Bluetooth 4.0
* OS: 
  * Linux sunshine-hp 3.19.0-33-generic #38~14.04.1-Ubuntu SMP Fri Nov 6 18:17:28 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

Any idea how to fix this without access to the peripheral's firmware?

sandeepmistry, 2015-11-28:
@manuelmitasch thanks for providing a detailed issue report!

From the HCI dump:

1) First connect, gets disconnecting right a way which is strange. Might be caused by https://github.com/sandeepmistry/node-bluetooth-hci-socket/commit/c5d7f8455a33d1da3ccd3915dc20a32f38c91182

2) The duplicate services discovery callbacks twice issue is a result of 1) and https://github.com/sandeepmistry/noble/issues/193 (which is still open).

3) Write after the first notify event, a disconnect event is received with reason "Connection Timeout"

4) As for the notifications, things look ok, then the peripheral disconnects. Not sure why ...

Let's start with 1), since there are not disconnect commands in your code snippet, I'm thinking the kernel is acting up, could you please try downgrading the version of ```node-bluetooth-hci-socket``` you are using to v0.4.0

```
2015-11-20 12:13:57.625667 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 00:07:80:79:0A:C9 type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6
2015-11-20 12:13:57.627540 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2015-11-20 12:14:00.722034 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role master
      bdaddr 00:07:80:79:0A:C9 (Public)

// ....

    
2015-11-20 12:14:00.827089 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection

// ....

2015-11-20 12:14:00.851416 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 00:07:80:79:0A:C9 type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6
2015-11-20 12:14:00.853050 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2015-11-20 12:14:00.975082 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
2015-11-20 12:14:01.632076 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role master
      bdaddr 00:07:80:79:0A:C9 (Public)

// ...

2015-11-20 12:14:18.068840 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x08
    Reason: Connection Timeout
2015-11-20 12:14:18.087372 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 00:07:80:79:0A:C9 type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6
2015-11-20 12:14:18.088838 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2015-11-20 12:14:18.216905 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role master
      bdaddr 00:07:80:79:0A:C9 (Public)

// ...

2015-11-20 12:14:29.552383 < HCI Command: Disconnect (0x01|0x0006) plen 3
[... 59 lines cut ...]
      handle 0x002b
      value 0x01 0x0d 0x02 0x05 0x00 0x02 0x00 0x00 0x00 0x00 
2015-11-20 12:14:24.330815 > ACL data: handle 64 flags 0x02 dlen 17
    ATT: Handle notify (0x1b)
      handle 0x002b
      value 0x02 0xac 0x00 0x06 0x00 0x02 0x00 0x00 0x00 0x00 
2015-11-20 12:14:25.245833 > ACL data: handle 64 flags 0x02 dlen 17
    ATT: Handle notify (0x1b)
      handle 0x002b
      value 0x01 0xf4 0x01 0x07 0x00 0x03 0x00 0x00 0x00 0x00 
2015-11-20 12:14:25.485891 > ACL data: handle 64 flags 0x02 dlen 17
    ATT: Handle notify (0x1b)
      handle 0x002b
      value 0x02 0x7a 0x00 0x08 0x00 0x03 0x00 0x00 0x00 0x00 
2015-11-20 12:14:28.576223 > ACL data: handle 64 flags 0x02 dlen 17
    ATT: Handle notify (0x1b)
      handle 0x002b
      value 0x03 0xb9 0x0b 0x09 0x00 0x03 0x00 0x00 0x00 0x00 

// ...

2015-11-20 12:14:29.552383 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
```

sandeepmistry, 2016-01-03:
@manuelmitasch any updates on this?

manuelmitasch, 2016-01-09:
@sandeepmistry Sorry, I did not have time to work on this project the last weeks. I will try to have a look next week and give you an update.

sandeepmistry, 2016-01-23:
Closing due to lack of activity, feel free to re-open with requested info.

closed 2016-01-23 by sandeepmistry

manuelmitasch, 2016-03-30:
Sorry for the inactivity. And thank you for this amazing library.
I'm back on the project. I do not see this issue appear anymore. 
Not sure whether it's because of a new firmware for the bluetooth device or a newer noble version.