noble/noble #274: Intermittent errors on Windows

mhle, 2015-11-20:
Sometimes getting:

```
Error: LIBUSB_TRANSFER_STALL
    at Error (native)(anonymous function) @ Logger.js:136errorHandler @ misc.js:128emitOne @ events.js:77emit @ events.js:169window.onerror 
```
and 
```
Error: LIBUSB_TRANSFER_STALL
    at Error (native)(anonymous function) @ Logger.js:136errorHandler @ misc.js:128emitOne @ events.js:77emit @ events.js:169window.onerror 
```

OS: Windows 10
Dongle: BCM20702A0 dongle with PID 21E8 and VID 0A5C
Platform: Electron

Communication with peripherals stop when encountering these errors. Is there a way to recover from these?

sandeepmistry, 2015-11-28:
@mhle have you looked it what USB endpoint is stalling?

Also, do you have code and steps to consistently reproduce?

mhle, 2015-12-08:
Looking into this further, this seems to be caused by too many hci events blocking communication with the dongle. We could consistently produce this by triggering a surge of data going to/from dongle by: 

1. trying to connect/disconnect to multiple (7+) devices at the same time,
2. continuously triggering startscan and stopscan with short intervals

sandeepmistry, 2016-01-03:
@mhle can we close this or are there any action items on the noble side?

closed 2016-01-03 by mhle