noble/noble #276: Error handling?

rclai, 2015-12-02:
I notice that every action called (writes in particular as well as others) assumes a successful outcome as seen by how the callback is always called with a null error object.

Is this a bluetooth rule of thumb that errors should never be emitted back?

sandeepmistry, 2015-12-05:
@rclai these are placeholders for now, errors still need to be bubbled up for some actions.

See https://github.com/sandeepmistry/noble/issues/184 and https://github.com/sandeepmistry/noble/issues/207.

tybro0103, 2017-03-10:
https://github.com/sandeepmistry/noble/pull/564

Bonitis, 2017-03-25:
@tybro0103 this may not be the best place for this; do you have any best practices for handling errors with noble? I'm dealing with a lot of errors in connections, service discovery, and characteristic discovery. This can usually be resolved by trying to re-connect to the device. Would it cause problems or be overly expensive to call `noble.startScanning()` whenever I hit an error in the connection/discovery process?