noble/noble #280: Reading Major and minor of a BLE device

karthikbnd, 2015-12-04:
Hello
I am able to run the noble and get the mac address, but i want to get the UUID, Major and Minor of a BLE(Beacon) device. How can i get that?

sandeepmistry, 2015-12-05:
@karthikbnd please see the following library: https://github.com/sandeepmistry/node-bleacon

closed 2015-12-05 by sandeepmistry

karthikbnd, 2015-12-07:
Hai
I want a test program to scan a beacon and console log the Major, minor and
rssi.
Can you please provide it.
I did the following but its at scanning
var Bleacon = require('bleacon');

//Bleacon.startScanning([uuid], [major], [minor]);// scan for bleacons with
a particular uuid. major, and minor
Bleacon.startScanning([]); // scan for any bleacons
console.log('Started Scanning........');
Bleacon.on('discover', function(bleacon) {
    console.log('beacon found: ' + JSON.stringify(bleacon));
    var uuid = bleacon.uuid;
    var major = bleacon.major;
    var minor = bleacon.minor;
    var date = new Date();
    console.log('Found BLE:', 'UUID:',uuid, ' Major:',major,'Minor:',minor,
'Date:',date );
    //var measuredPower = manufacturerData.readInt8(24);
});

Bleacon.stopScanning();

output is
y@hubblue:~/node_modules$ node filescan_beacon.js
Started Scanning........

sandeepmistry, 2015-12-08:
@karthikbnd you will need to run with ```sudo``` or run as root.

Alternatively you can following these instructions: https://github.com/sandeepmistry/noble#running-without-rootsudo

karthikbnd, 2015-12-14:
Hai
I want to scan for every 10sec and get the data, can i get help in setting
the interval

cyclonmaster, 2017-05-06:
I follow the node-bleacon same as karthik did, also get nothing from estimote:

pi@raspberrypi:~/Downloads $ sudo node bleaconscan.js
Started Scanning........

Do I need to configure anything?

karthikbnd, 2018-11-21:
Hallo,

ich habe gerade die Petition „sushma swaraj : Long term visa for dependant
parents of Indians in Germany" unterschrieben und wollte dich fragen, ob du
auch mitmachst.

Unser Ziel ist es, 781 Unterschriften zu sammeln und dafür brauchen wir
Unterstützung. Hier kannst du mehr über die Petition erfahren:

http://chng.it/dZfpvd2XhV

Vielen Dank!
Karthik