noble/noble #282: Are Node.js, Zadig and noble not sufficient?

boredSinan, 2015-12-10:
Are the prerequisites (https://github.com/sandeepmistry/noble#windows) really necessary?

I started with Noble a few weeks ago. I installed every prerequisite on my Windows 8.1 computer (PC1).
I am implementing a tracking system application which is based on the example "enter-exit.js".

There are about 20 other PCs I want to install my application to. 
First I assumed that I have to do the whole prerequisites-procedure on all the PCs!!
That would be terrible!!

But today I installed on one of the PCs (PC2) only the following tools, and I could run my application:
- Zadig tool for WinUSB driver (Dongle CSR8510A10 is plugged in)
- Node.js
- Last, I copied the "node-modules" folder (which was created on PC1) to PC2.

That's it! 
The following tools aren't installed on PC2:
- Python 2.7
- MS VS 2013 Express (or any other VS)
- I didn't run any command such as (npm install node-gyp, npm install bluetooth-hci-scoket)

I didn't spend so much time with the noble project, therefore I don't know how it works exactly?

My Questions:
1) Are Node.js, Zadig and the "node-modules" folder sufficient to run any application which is based on noble?
2) Why exactly are Python and the C++ compiler in VS2013 needed for?
3) Do I really need the installation of node-gyp and bluetooth-hci-socket?

taktran, 2015-12-10:
I think those tools are needed to build noble on windows (ie, to run `npm install`) only. After you build it, you can use the built files (ie, `node_modules/noble`) on similar platforms.

Also Python is required for node to be installed, and the C++ compiler is probably used to build the native modules.

boredSinan, 2015-12-10:
Hi taktran,
I could imagine the following:
The JavaScript codes will be interpreted by Node.js everytime I run my application, so there are no compiled/built files (at least for JavaScript and Python). But there should be compiled/built files for C++.

Maybe, the first time, I installed the prerequisites on PC1, all the C++ files had been built and generated! Therefore I could easily copy&paste the "node-modules" folder to PC2 and run my app?

sandeepmistry, 2015-12-12:
@boredSinan 

> 1) Are Node.js, Zadig and the "node-modules" folder sufficient to run any application which is based on noble?

Yes, as long as the same version of Node.js is being used, as well as same CPU arch (32-bit vs 64-bit).

> 2) Why exactly are Python and the C++ compiler in VS2013 needed for?

See @taktran's answer, and [node-gyp](https://github.com/nodejs/node-gyp) project.

3) Do I really need the installation of node-gyp and bluetooth-chi-socket?

```bluetooth-hci-socket``` could be split up into 2 separate modules: ```bluetooth-hci-socket-linux``` and ```bluetooth-hci-socket-usb``` .```bluetooth-chi-socket-usb``` would depend on ```node-usb```.

Another option is to use [node-pre-gyp](https://www.npmjs.com/package/node-pre-gyp) and have the Node.js native module pre-built for certain platforms. However, then you have to decide on the version of Node.js you want to pre-build for, 32-bit vs 64-bit, etc (a big compatibility matrix).

sandeepmistry, 2016-01-03:
@boredSinan can we close this or are there follow on action items?

closed 2016-01-06 by boredSinan