noble/noble #283: Connect fails on Raspberry

paulgavrikov, 2015-12-10:
Hi,

again thank you for your project. While my noble app works perfectly on my Mac, connections fail on my RPi.

My code is fairly simple:
```
function access (peripheral) {
     // this is reached
    peripheral.connect(function(err) {
      .... // never reached
```
Here is the interesting part of my hci log:

```
2015-11-22 18:36:54.732907 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 00:12:6F:6B:C7:32 type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6
2015-11-22 18:36:54.734511 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x0c ncmd 1
    Error: Command Disallowed
```

Apparantly it just diconnects - any ideas why or how to resolve this?

I am on 
```
Linux raspberrypi 4.1.13+ #826 PREEMPT Fri Nov 13 20:13:22 GMT 2015 armv6l GNU/Linux
````

bleno & noble are stock - non modified.

sandeepmistry, 2015-12-12:
@paulgavrikov could you please share the full code running and full HCI dump?

Are you using noble and bleno together in the same app?

paulgavrikov, 2015-12-13:
Thanks for your reply. Yes I am using both noble and bleno. Let me quickly explain my scenario:
Smartphone connects to Raspberry (bleno) and reads a characteristic value.
Raspberry (bleno) returns a value and starts discovery via (noble) to connect to a third device. This works on OS X but on Raspbian - even if I disconnect before the scan from my smartphone - it won't connect to the third device (gets stuck, hcidump shows error)

IF for some reason it is not possible to run bleno & noble in Linux parallel - could you add a possibility to select which hci interface noble & bleno should use? Then I would use two separate dongles!

index.js
```
var bleno = require('bleno');

var BleService = bleno.PrimaryService;
var LogChar = require('./char');

bleno.on('stateChange', function(state) {
  console.log('on -> stateChange: ' + state);

  if (state === 'poweredOn') {
    bleno.startAdvertising("HELLO", []);
  } else {
    bleno.stopAdvertising();
  }
});

bleno.on('advertisingStart', function(error) {
  console.log('on -> advertisingStart: ' + (error ? 'error ' + error : 'success'));

  if (!error) {

    bleno.setServices([
 
      new BleService({
    	uuid: '180a',
    	characteristics: [
	    	new LogChar('180a','2a25','read')
    	]
    	})
    ]);

  } else {
    bleno.stopAdvertising();
    bleno.startAdvertising("HELLO", []);
  }
});

```

char.js
```
var util = require('util');

var bleno = require('bleno');
var noble = require('noble');

var BlenoCharacteristic = bleno.Characteristic;

var _device_uuid = '00126f6bc732';

var that;

function LogChar(serviceUUID, uuid, props) {

  this.serviceUUID = serviceUUID;

  BlenoCharacteristic.call(this, {
    uuid: uuid,
    properties: props.split("|")
  });
};

util.inherits(LogChar, BlenoCharacteristic);
[... 130 lines cut ...]
2015-12-13 13:41:08.514737 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Data (0x08|0x0008) ncmd 1
    status 0x00
2015-12-13 13:41:08.516841 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x00
2015-12-13 13:41:08.521744 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Response Data (0x08|0x0009) ncmd 1
    status 0x00
2015-12-13 13:41:08.526744 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Data (0x08|0x0008) ncmd 1
    status 0x00

HERE WE CONNECT FROM SMARTPHONE

2015-12-13 13:41:43.209812 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2015-12-13 13:41:46.574224 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2015-12-13 13:41:46.597220 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x0c ncmd 1
    Error: Command Disallowed
```

paulgavrikov, 2015-12-13:
Okay, I found a mailing-list: http://sourceforge.net/p/bluez/mailman/message/15799497/
It states that CSR Chips don't allow multiple connection attempts. However I have the same behaviour with a Broadcom dongle. My Mac uses the built in Broadcom 20702A3 which works and seems to support parallel connections.

If I disconnect before the scan and wait a few seconds it works - but I need parallel connections for my app.
Any chance to make noble & bleno support multiple hci interfaces?

sandeepmistry, 2015-12-13:
@paulgavrikov based on https://github.com/sandeepmistry/bleno/issues/158 did you get this working by setting the relevant environment variables to specify the HCI interface?

paulgavrikov, 2015-12-14:
> @paulgavrikov based on sandeepmistry/bleno#158 did you get this working by setting the relevant environment variables to specify the HCI interface?

Yes, using 2 separate hci interfaces works - but could it also work with one ?

sandeepmistry, 2015-12-20:
@paulgavrikov only if you have a Bluetooth 4.1 adapter. There's a change in noble master that reports the command disallowed connect failure.

closed 2015-12-20 by sandeepmistry