noble/noble #287: "hciconfig -a" resetting noble state.

tasasaki-github, 2015-12-17:
While scanning BLE packets with noble, executing "hciconfig -a" command resets noble state but it shouldn't.
We can reproduce the problem with the following code:

```javascript
// start scanning BLE packets
noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    noble.startScanning([], true);
  }
  else {
    noble.stopScanning();
  }
});
```

When we execute "hciconfig -a" command after noble.startScanning(), state change occurs and noble.stopScanning called wrongly.

sandeepmistry, 2016-01-01:
Closed via https://github.com/sandeepmistry/noble/commit/8d6df4f075ec5fe281eba3181da17e11f2a1179f

You can try out the change via ```npm install sandeepmistry/noble``` before a new release is published to npm.

closed 2016-01-01 by sandeepmistry

tasasaki-github, 2016-01-05:
Thank you. I confirmed the unnecessary state change event now never occurred by executing hciconfig -a command.

sandeepmistry, 2016-01-05:
@tasasaki-github great, thank you for verifying! A new version with the changes should be published to npm later this week.