noble/noble #288: Enhancement for getting the local bluetooth device address.

tasasaki-github, 2015-12-17:
The local bluetooth device address is of the same importance as the peripheral addresses on many applications.

For example, on some kind of indoor positioning applications, the scanner's address and peripherals' addresses and RSSIs are used as basic information for locating peripherals.

But now we have to get the local address like the following way:
```javascript
var myMac;
noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    // get this scanner's mac address.
    myMac = noble._bindings._hci.address;
  }
});
```
_bindings and _hci are for internal use, we shouldn't access them directly.
So it's better to make an api to get the local address for application users.

boredSinan, 2015-12-17:
I am having the same issue and found out that the following "usb" library is helpful but I still have a problem:

Link: https://www.npmjs.com/package/usb

```javascript
var usb = require('usb');
console.log(usb.getDeviceList());

//Find the first device with the specified ids:
Device = usb.findByIds(2578, 1);
//console.log(Device.deviceDescriptor);
//console.log(Device.configDescriptor);
```

##### My problem:
I wanna get the devices unique address (e.g the MAC address).  I plugged in 2 BLE USB dongles of the same manufacturer, and the output is as follows:

[image]

-> The "busNumber" is the USB slot number in which the dongle is plugged in.
-> I don't understand exactly what "deviceAddress" stands for and why it is only a single digit!

I assume that the above mentioned library doesn't differ between 2 devices of the same manufacturer. Here are my questions:
1) Can you change the idVendor and idProduct with the "usb" library?
2) Does the "usb" library provide the devices unique address?
3) Are there any riscs how "tasasaki-github" is getting the hci-address?

sandeepmistry, 2016-01-01:
```noble.address``` now contains local Bluetooth adapter address after ```stateChange``` event (https://github.com/sandeepmistry/noble/commit/cbe0213e5f9d02ce40e8873a44f1f9008c0d4397).

You can try out the change via ```npm install sandeepmistry/noble``` before a new release is published to npm.

@boredSinan:
> 2) Does the "usb" library provide the devices unique address?

USB device address is different from Bluetooth adapter address.

> 3) Are there any riscs how "tasasaki-github" is getting the chi-address?

Yes, it's reaching into noble internals, which could change in the future. Also, if used on OS X it would not work.

closed 2016-01-01 by sandeepmistry

tasasaki-github, 2016-01-05:
Thank you. I confirmed I could get the local address via noble.address.
I didn't know the previous way didn't work on OS X.

sandeepmistry, 2016-01-05:
@tasasaki-github great, thank you for verifying! A new version should with the change should be published to npm later this week.