noble/noble #292: Cannot scan for advertising peripheral in background

arusu0629, 2015-12-22:
Peripheral : my ios application for swift
Central : Raspberry Pi using noble

I want to scan my ios application as peripheral in background with noble.
But, cannot find it.

Instead of using noble, I use lightBlue as central, works well.
I'm sorry for not good at explaining in English.
How can I solve it?

```swift
var noble = require('noble');

var scanUUID = '66995c84a3294112b745c997475cf017';
var scanUUIDs = [scanUUID];

noble.on('stateChange', function(state) {
    console.log('on -> stateChange: ' + state);

    if (state == 'poweredOn') {
        noble.startScanning(scanUUIDs, false);
    } else {
        noble.stopScanning();
    }
});

noble.on('scanStart', function() {
    console.log('on -> scanStart');
});

noble.on('discover', function(peripheral) {
    var name = peripheral.advertisement.localName; // maybe ignored in background
    var uuid = peripheral.advertisement.serviceUuids[0];
    console.log('peripheral serviceUUID = ' + uuid);
    console.log('peripheral name = ' + name);
});
```

sandeepmistry, 2015-12-23:
@arusu0629 have you tried it with an Android central as well? What happens if you try to scan for all peripherals instead of a specific UUID?

I'm wondering if there is something special about iOS central and iOS peripheral.

arusu0629, 2015-12-24:
@sandeepmistry Thank you for replying my question.
I figure out my problem reading [this](https://developer.apple.com/library/ios/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW1)

#### The bluetooth-peripheral Background Execution Mode
All service UUIDs contained in the value of the CBAdvertisementDataServiceUUIDsKey advertisement key are placed in a special “overflow” area; they can be discovered only by an iOS device that is explicitly scanning for them.

So, cannot scan for advertising peripheral in background.
I'm sorry for lacking my wisdom.

By the way, I ask you another question.
How do you decide to peripheral variable "id" ?

```
peripheral = {
  id: "<id>", <- this one
  address: "<BT address">, // Bluetooth Address of device, or 'unknown' if not known
  addressType: "<BT address type>", // Bluetooth Address type (public, random), or 'unknown' if not known
```

sandeepmistry, 2016-01-03:
@arusu0629 no worries.

> How do you decide to peripheral variable "id" ?

 * Linux: it's the Bluetooth Address without the ```:```'s
 * OS X: id provided by CoreBluetooth.

closed 2016-01-03 by sandeepmistry

arusu0629, 2016-01-07:
@sandeepmistry 
Thank you for answering my question.