noble/noble #293: Cannot receive notification packets successfully when the number of packets gets big

lijunxyz, 2015-12-29:
I want to transfer some data from a BLE peripheral (a sensor with on-board external flash memory) to a BLE central (Raspberry Pi + BLE USB dongle + noble). The model is quite simple: I write a command to a characteristic on peripheral, and then get a series of notification packets containing the sensor data back. Each notification packet has 18 bytes of data, and it will take a few thousands of packets to transfer all data. On the peripheral side, I set the interval of notification to 10~20 ms. Below is a related code snippet showing my implementation: 

```
      var notif_count = 0;
      ......
      var characteristic = some_characteristic_object_of_interest;
      characteristic.on('data', function (data, isNotif) {  // register handler
        if (isNotif == true) {
          console.log("Notification count = " + (++notif_count));
          console.log("Characteristics " + characteristic.uuid + " notification data: ");
          console.log(data);
        }
        else {
          console.log("Characteristics " + characteristic.uuid + "is false");
        }
      });
      characteristic.notify(true, function (err) {  // turn on CCCD
        console.log("Notif of characteristics " + characteristic.uuid + " is on");
      });
      ......
```

This works well if the packet number is small (like <100) and the transfer finishes quickly. However, as long as the number exceeds a few hundreds, it will fail and then disconnect. The number of packets for it to fail varies among sessions. To make it work, I have to set the notification interval very large (>100 ms) on the peripheral side at the cost of data transfer rate, which is not desirable. 

I also test the same peripheral (short notification interval) with another BLE central app (Nordic Master Control Panel PC app by manually writing to the characteristic to initiate the transfer, and it finishes with receiving thousands of notifications successfully. So I think the peripheral is just fine, and there must be something wrong with my Nodejs script. 

Any thoughts on this matter will be appreciated.

sandeepmistry, 2016-01-03:
@lijunxyz this might be connection interval related.

Please provide a HCI dump capture: ```sudo hcidump -t -x```

lijunxyz, 2016-01-04:
The hicdump log is [here](https://dl.dropboxusercontent.com/u/9230210/hcidump_log.txt).

sandeepmistry, 2016-01-05:
@lijunxyz thanks for capturing, here are some sections of interest:

1) I'm not sure what the initial connection interval negotiated is (not shown in the capture - LE Connection Complete).

2) At 2016-01-03 22:36:52.252725 a LE Connection Update is sent. I don't think this is noble, but something inside the Linux kernel.

3) At 2016-01-03 22:36:52.535851 a connection interval of 200 ms is selected.

4) At 2016-01-03 22:36:52.956915 the peripheral sends a disconnect.

Overall, after 3) you probably pumping out more data than can be buffered so the peripheral disconnects. A connection interval of 200 ms beens each side syncs to exchange data every 200.0ms.

What kernel version are you using? (```uname -a```)

```
2016-01-03 22:36:31.499785 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 94:54:93:02:98:17 type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6

...

2016-01-03 22:36:32.253821 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 68, role master
      bdaddr 94:54:93:02:98:17 (Public)

... 
2016-01-03 22:36:31.499785 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 94:54:93:02:98:17 type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6

...

2016-01-03 22:36:32.253821 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 68, role master
      bdaddr 94:54:93:02:98:17 (Public)

... 

2016-01-03 22:36:52.252725 < HCI Command: LE Connection Update (0x08|0x0013) plen 14
  44 00 28 00 A0 00 00 00 90 01 00 00 00 00 
2016-01-03 22:36:52.256858 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1

...

2016-01-03 22:36:52.535851 > HCI Event: LE Meta Event (0x3e) plen 10
    LE Connection Update Complete
      status 0x00 handle 68
      interval 200.00ms, latency 0.00ms, superv. timeout 4000.00ms
2016-01-03 22:36:52.553876 > ACL data: handle 68 flags 0x02 dlen 26
    ATT: Handle notify (0x1b)
      handle 0x0019
      value 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x10 0xe2 0x05 
2016-01-03 22:36:52.746504 > ACL data: handle 68 flags 0x02 dlen 26
    ATT: Handle notify (0x1b)
      handle 0x0019
      value 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x00 0xe2 0x05 
2016-01-03 22:36:52.747933 > ACL data: handle 68 flags 0x02 dlen 26
    ATT: Handle notify (0x1b)
      handle 0x0019
      value 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xf0 0xe1 0x05 
2016-01-03 22:36:52.748142 > ACL data: handle 68 flags 0x02 dlen 26
    ATT: Handle notify (0x1b)
      handle 0x0019
      value 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xe0 0xe1 0x05 
2016-01-03 22:36:52.748256 > ACL data: handle 68 flags 0x02 dlen 26
    ATT: Handle notify (0x1b)
      handle 0x0019
      value 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xd0 0xe1 0x05 
2016-01-03 22:36:52.947153 > ACL data: handle 68 flags 0x02 dlen 26
    ATT: Handle notify (0x1b)
      handle 0x0019
      value 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xc0 0xe1 0x05 
2016-01-03 22:36:52.956915 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 68 reason 0x13
    Reason: Remote User Terminated Connection
```

lijunxyz, 2016-01-12:
I am using it on a Raspberry Pi. The output of `uname -a` is

```
Linux drpi6 4.1.7+ #817 PREEMPT Sat Sep 19 15:25:36 BST 2015 armv6l GNU/Linux
```

From what you described, the problem may be from the long connection interval 200ms which is the maximum connection interval set by the firmware (while the minimum connection interval is 50 ms). 

It will be great if I can specify the connection interval in noble. It was briefly discussed in issue #270. Could you elaborate a little more?

vshymanskyy, 2016-01-13:
I think I'm having a related trouble.
I also don't know how to determine the delay time between notifications. Any hints?
Thanks!

sandeepmistry, 2016-01-17:
The root cause of the issue is the Linux kernel sending the LE connection update:

```
2016-01-03 22:36:52.252725 < HCI Command: LE Connection Update (0x08|0x0013) plen 14
  44 00 28 00 A0 00 00 00 90 01 00 00 00 00 
```

The only way to get it to stop interfering is to switch to user channel HCI socket mode:
https://github.com/sandeepmistry/node-bluetooth-hci-socket#user-channel

I've been holding off on this change because it's more awkward for users. The HCI adapter needs to be powered down via ```sudo hciconfig hci0 down``` prior to starting noble.

You can try out this branch: https://github.com/sandeepmistry/noble/tree/bind-user
```sh
sudo HCI_CHANNEL_USER=1 node <file>.js
```

lijunxyz, 2016-01-18:
@sandeepmistry I first powered down BLE adapter using `sudo hciconfig hci0 down`, and then ran `sudo HCI_CHANNEL_USER=1 node <file>.js`. But there were no output messages, probably because it was not running. It seemed HCI adapter was not turned back on by the NodeJS script. Is this expected?

sandeepmistry, 2016-01-23:
@lijunxyz are you using the the ```bind-user``` branch I linked to?

lijunxyz, 2016-01-23:
@sandeepmistry Yes

sandeepmistry, 2016-01-30:
Have you looked at what's going on via the ```hcidump``` command?

sandeepmistry, 2016-02-21:
Closing for now due to lack of activity, please re-open with requested info if you are still interested.

closed 2016-02-21 by sandeepmistry

jeberhardt, 2016-04-20:
Is using bind-user faster?

sandeepmistry, 2016-04-24:
@jeberhardt 

> Is using bind-user faster?

It should roughly be the same. The kernel doesn't process HCI adapter events.

PanzerHabba, 2016-08-24:
Hi @sandeepmistry 

Great work done on node and BLE!

I have the same problem whcih @lijunhw  adressed. I want to send files > 1Mbyte, which means a lot of packages. I tried to install the node-bluetooth-hci-socket package but didn`t have any luck running it. trying: `sudo HCI_CHANNEL_USER=1 node myScript.js` resulted in:
`buffer.js:564
        return this.hexWrite(string, offset, length);

        TypeError: Invalid hex string`

Does install the node-bluetooth-hci-socket work with Rpi3 internal BLE-chip ?
Is there no other way to prevent the Linux kernel sending the LE connection update ?

There is offcourse a possibility that I don`t understand how to use node-bluetooth-hci-socket (not much documentation available :))

PanzerHabba, 2016-08-25:
Also when I use the bind-user branch and try:
`sudo DEBUG=* HCI_CHANNEL_USER=1 node myapp.js`
I get the following output:
`noble stateChange poweredOff +0ms`

So it seems like it wont power up the hci adapter. Also the hcidump gave nothing:
`HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff`

PanzerHabba, 2016-08-26:
I have been able to found a workaround to the LE connection update:

`var exec = require('child_process').exec;
var child;

child = exec('sudo hcitool lecup --handle 64 --min 12 --max 15', function(error, stdout, stderr) {
            if (error) {
                console.log(error);
                return;
            }
        });
`

But this will offcourse rely on knowing the connection handle. As I would imagine this could change from time to time ?

I`m calling the exec function during the subscribe callback to one of the characteristics.
This is not a pretty solution, but it allow me to increase the throughput a lot.

sandeepmistry, 2016-10-10:
@PanzerHabba thanks for sharing your workaround, however I think this is a issue with the peripheral's configuration.

```
2016-01-03 22:36:52.252483 > ACL data: handle 68 flags 0x02 dlen 16
    L2CAP(d): cid 0x0005 len 12 [psm 0]
      12 02 08 00 28 00 A0 00 00 00 90 01 
2016-01-03 22:36:52.252655 < ACL data: handle 68 flags 0x00 dlen 10
    L2CAP(d): cid 0x0005 len 6 [psm 0]
      13 02 02 00 00 00 
2016-01-03 22:36:52.252725 < HCI Command: LE Connection Update (0x08|0x0013) plen 14
  44 00 28 00 A0 00 00 00 90 01 00 00 00 00 
2016-01-03 22:36:52.256858 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 cmd 1
```

```
12 02 08 00 
28 00 - min latency 50 ms
A0 00 - max latency 200 ms
00 00 - latency 0ms
90 01 - supervision timeout 4000ms
```

So the Linux kernel is respecting the devices connection parameter update request.

rabee05, 2020-11-22:
> I have been able to found a workaround to the LE connection update:
>
> `var exec = require('child_process').exec;`
> `var child;`
>
> `child = exec('sudo hcitool lecup --handle 64 --min 12 --max 15', [...]

Thank you for the workaround  - I am not able to use it. May I need to use different handle? What is the connection handle and how to find it?