noble/noble #294: HCI Socket never returns data - hcidump shows data though

JohannesRudolph, 2015-12-30:
I'm not sure where the problem is here, it may well be a node issue (running this on OpenWRT, node 12.7).

I'm starting the advertisement-discovery.js sample in one console:
```
root@host:/rim# DEBUG=* node noble-test.js 
  hci setting filter to: 1400000020c10000000000400000 +0ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +99ms
  hci set le event mask - writing: 010120081f00000000000000 +48ms
  hci read local version - writing: 01011000 +9ms
  hci write LE host supported - writing: 016d0c020100 +11ms
  hci read LE host supported - writing: 016c0c00 +7ms
  hci read bd addr - writing: 01091000 +7ms```
```
and there it hangs. The ```onStateChange``` event is never fired. Apparently, ```Hci.prototype.onSocketData``` is never fired (no output in debug log). 

Even though, my BTLE USB device correctly sends command complete events, output from a second console: 
```
root@host:~# hcidump 
HCI sniffer - Bluetooth packet analyzer ver 5.30
device: hci0 snap_len: 1500 filter: 0xffffffff
< HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
> HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
< HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
< HCI Command: Read Local Version Information (0x04|0x0001) plen 0
> HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x1000
    LMP Version: 4.0 (0x6) LMP Subversion: 0x220e
    Manufacturer: Broadcom Corporation (15)
< HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
> HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
< HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
> HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
< HCI Command: Read BD ADDR (0x04|0x0009) plen 0
> HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 5C:F3:70:69:50:00
```

Any idea what's going wrong here? Why does the HCI socket never return any data?

sandeepmistry, 2016-01-03:
@JohannesRudolph what kind of processor does your OpenWRT device have? If it's MIPS, maybe it's an Big vs Little Endian issue.

You could try setting the HCI socket filter mask to ```0xffffffff``` (```eventMask1``` and ```eventMask2```) like ```hcidump``` does here:

https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/hci.js#L125

JohannesRudolph, 2016-01-03:
Yes, it's a MIPS24k with Big Endian. How does that relate to noble?

sandeepmistry, 2016-01-03:
@JohannesRudolph see my previous comment about the filter mask.

I'm thinking the endianness is incorrect when setting the filter from a Big Endian based system.

http://lxr.free-electrons.com/source/include/net/bluetooth/hci_sock.h#L49

The Node.js side is currently assuming Little Endian, whereas is the struct linked above was used in C/C++ it would be set accordingly. Please try my suggestion of setting the masks to ```0xffffffff``` then we can take it from there for a more suitable fix.

sandeepmistry, 2016-01-23:
Closing due to lack of activity, feel free to re-open with requested info from my last comment.

closed 2016-01-23 by sandeepmistry

JohannesRudolph, 2016-02-21:
Hi @sandeepmistry , sorry for the lack of response, I had to put this on hold. 

So I tried your suggestion and changed the filter masks as follows: 
```
  var filter = new Buffer(14);                                                     
  var typeMask = (1 << HCI_EVENT_PKT)| (1 << HCI_ACLDATA_PKT);                 
  var eventMask1 = 0xffffffff; //(1 << EVT_DISCONN_COMPLETE) | (1 << EVT_ENCRYPT_CHANGE) | (1 << EVT_CMD_COMPLETE) | (1 << EVT_CMD_STATUS);
  var eventMask2 = 0xffffffff; //(1 << (EVT_LE_META_EVENT - 32));                                                                          
  var opcode = 0;                                                                                                                          
                                                                               
  filter.writeUInt32LE(typeMask, 0);                                                                         
  filter.writeUInt32LE(eventMask1, 4);                                                                                                     
  filter.writeUInt32LE(eventMask2, 8);                                                                                                     
  filter.writeUInt16LE(opcode, 12);          
````
With that I get the following output (still not getting a poweredOn stateChange event): 

```
$ DEBUG=* node test.js
  hci setting filter to: 14000000ffffffffffffffff0000 +0ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +144ms
  hci set le event mask - writing: 010120081f00000000000000 +88ms
  hci read local version - writing: 01011000 +10ms
  hci write LE host supported - writing: 016d0c020100 +6ms
  hci read LE host supported - writing: 016c0c00 +7ms
  hci read bd addr - writing: 01091000 +7ms
```

Any idea how to take it further from here? Using noble 1.2.1.

sandeepmistry, 2016-02-21:
@JohannesRudolph try also setting ```typeMask``` to  0xffffffff.

JohannesRudolph, 2016-06-13:
Sorry for the long delay in replying. Setting typeMask to `0xffffffff` did the trick. Can you come up with a fix based on this? I assume checking platform endianness and then inverting or replacing `writeUIntXXLE` with `writeUIntXXBE` should do the trick.

I can work on a PR, probably within the next few days but we'd need an updated bleno&noble release from you.

sandeepmistry, 2016-06-20:
@JohannesRudolph I mentioned this to @marcnicholas a while back. He didn't have any issues with the mask on his MIPS system.

However, let me know if you have a PR to calculate the mask based on the endianness of the system.

marcnicholas, 2016-06-20:
I was using a MIPS in Little Endian mode. (mipsel)

sandeepmistry, 2016-06-20:
@marcnicholas thanks for the clarification, this makes more sense :)