noble/noble #295: Crypto error when connecting to WICED Sense

deadprogram, 2016-01-03:
This is to replace https://github.com/sandeepmistry/node-bluetooth-hci-socket/issues/15

When connecting to the Broadcom WICED Sense tag from Linux (Ubuntu 14.04), I hit the following error. I do not have this problem with any other BLE hardware:

```
buffer.js:229
      length += list[i].length;
                       ^

TypeError: Cannot read property 'length' of undefined
    at Function.Buffer.concat (buffer.js:229:24)
    at Object.c1 (/home/ron/Development/cylonjs/cylon-example/node_modules/cylon-ble/node_modules/noble/lib/hci-socket/crypto.js:8:19)
    at Smp.handlePairingResponse (/home/ron/Development/cylonjs/cylon-example/node_modules/cylon-ble/node_modules/noble/lib/hci-socket/smp.js:86:12)
    at Smp.onAclStreamData (/home/ron/Development/cylonjs/cylon-example/node_modules/cylon-ble/node_modules/noble/lib/hci-socket/smp.js:57:10)
    at emitTwo (events.js:92:20)
    at emit (events.js:172:7)
    at AclStream.push (/home/ron/Development/cylonjs/cylon-example/node_modules/cylon-ble/node_modules/noble/lib/hci-socket/acl-stream.js:35:10)
    at NobleBindings.onAclDataPkt (/home/ron/Development/cylonjs/cylon-example/node_modules/cylon-ble/node_modules/noble/lib/hci-socket/bindings.js:252:15)
    at emitThree (events.js:97:13)
    at emit (events.js:175:7)
```

sandeepmistry, 2016-02-07:
@deadprogram as mentioned in https://github.com/sandeepmistry/node-bluetooth-hci-socket/issues/15   "As well as include what you've done so far to debug the issue ..."

It's really difficult for me to support you with out more info from you.

deadprogram, 2016-02-07:
That error was all I had at that point. Let me locate my WICED Sense and see if I can get additional details. Any suggestions on how to output some useful debug info?

sandeepmistry, 2016-02-07:
@deadprogram HCI dump would be useful.

As well as you spending sometime to better understand what was going wrong, etc.

deadprogram, 2016-02-07:
Hi, @sandeepmistry I have some further details. Note that this device was working with the versions of Noble from before the hci-socket refactor.

I am attempting to run the `examples/peripheral-explorer.js` program that is included with Noble with one of my WICED Sense tags. The following error occurs:

```
$ node node_modules/cylon-ble/node_modules/noble/examples/peripheral-explorer.js 207377654321
peripheral with ID 207377654321 found
  Local Name        = WICED Sense Kit
  TX Power Level    = 4
  Manufacturer Data = 0f000202f202203a
  Service Data      = 
  Service UUIDs     = 

services and characteristics:
buffer.js:229
      length += list[i].length;
                       ^

TypeError: Cannot read property 'length' of undefined
    at Function.Buffer.concat (buffer.js:229:24)
    at Object.c1 (/home/ron/Development/cylonjs/cylon-example/node_modules/cylon-ble/node_modules/noble/lib/hci-socket/crypto.js:8:19)
    at Smp.handlePairingResponse (/home/ron/Development/cylonjs/cylon-example/node_modules/cylon-ble/node_modules/noble/lib/hci-socket/smp.js:86:12)
    at Smp.onAclStreamData (/home/ron/Development/cylonjs/cylon-example/node_modules/cylon-ble/node_modules/noble/lib/hci-socket/smp.js:57:10)
    at emitTwo (events.js:92:20)
    at emit (events.js:172:7)
    at AclStream.push (/home/ron/Development/cylonjs/cylon-example/node_modules/cylon-ble/node_modules/noble/lib/hci-socket/acl-stream.js:35:10)
    at NobleBindings.onAclDataPkt (/home/ron/Development/cylonjs/cylon-example/node_modules/cylon-ble/node_modules/noble/lib/hci-socket/bindings.js:263:15)
    at emitThree (events.js:97:13)
    at emit (events.js:175:7)
```

Here is the associated info from the hcidump running at the same time that the error occurs:

```
$ hcidump
HCI sniffer - Bluetooth packet analyzer ver 2.5
device: hci0 snap_len: 1500 filter: 0xffffffffffffffff
> HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x1000
    LMP Version: 4.0 (0x6) LMP Subversion: 0x1000
    Manufacturer: Intel Corp. (2)
> HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
> HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
> HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 60:57:18:EE:05:74
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 2
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 2
    status 0x00
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 2
> HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 2
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
> HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 3585 reason 0x16
    Reason: Connection Terminated by Local Host
```

dethredic, 2016-02-17:
I'm running into this problem as well. I modified one of the examples to look like:
```
noble.on('discover', function(peripheral) {
  var advertisement = peripheral.advertisement;

  var localName = advertisement.localName;
  if (localName) {
    console.log('Local Name: ' + localName);
  }

  if (localName == "<name>") {
    peripheral.connect(function(error) {
      var serviceUUIDs = ["<service>"];
      var characteristicUUIDs = ["<uuid>"];
      peripheral.discoverSomeServicesAndCharacteristics(serviceUUIDs, characteristicUUIDs, function(error, services, characteristics) {
        var characteristic = characteristics[0];
        characteristic.read(function(error, data) {
        });
      });
    });
    return;
  }
});
```

```
buffer.js:491
      length += buf.length;
                   ^
TypeError: Cannot read property 'length' of undefined
    at Function.Buffer.concat (buffer.js:491:20)
    at Object.c1 (/home/phil/tmp/noble/lib/hci-socket/crypto.js:9:19)
    at Smp.handlePairingResponse (/home/phil/tmp/noble/lib/hci-socket/smp.js:86:12)
    at Smp.onAclStreamData (/home/phil/tmp/noble/lib/hci-socket/smp.js:57:10)
    at EventEmitter.emit (events.js:117:20)
    at AclStream.push (/home/phil/tmp/noble/lib/hci-socket/acl-stream.js:35:10)
    at NobleBindings.onAclDataPkt (/home/phil/tmp/noble/lib/hci-socket/bindings.js:272:15)
    at EventEmitter.emit (events.js:106:17)
    at Hci.onSocketData (/home/phil/tmp/noble/lib/hci-socket/hci.js:447:14)
    at BluetoothHciSocket.EventEmitter.emit (events.js:95:17)
```

A quick look makes it seem like `preq` is not getting initialized. It looks like the expected flow is: send pairing request (initialize `preq`) -> handle pairing response. In my case when I read the characteristic it triggers a security request (0x0B which seems to be unhandled by the module) and the first thing that the `smp.js` module sees is the pairing response (so `preq` isn't initialized).

Let me know if I can provide more information

sandeepmistry, 2016-02-21:
@dethredic debug output and HCI dump capture would be great! Thanks for sharing details on what you've debugged and diagnosed so far!

Debug output
```
sudo DEBUG=* node <file>.js
```

HCI Dump:
```
sudo hcidump -t -x
```

I'm wondering if the kernel is interfering with things ....

deadprogram, 2016-02-21:
I provided my own hci dump previously in https://github.com/sandeepmistry/noble/issues/295#issuecomment-181134959, @sandeepmistry but here is a gist with the associated debug output:

https://gist.github.com/deadprogram/4b69193dd573b71c0278

dethredic, 2016-02-22:
I modified smp.js a bit to handle the security request message (this also initializes `preq` for when `handlePairingResponse` gets called which sidesteps the crash):
```
...
 
var SMP_SECURITY_REQUEST = 0x0B;
 
...
 
Smp.prototype.onAclStreamData = function(cid, data) {
   ...
  } else if (SMP_SECURITY_REQUEST == code) {
    this.handleSecurityRequest(data);
  }
  ...
};
 
...
 
Smp.prototype.handleSecurityRequest = function(data) {
  console.log('handleSecurityRequest');
  var hexdump = require('hexdump-nodejs');
  console.log(hexdump(data));
 
  this._preq = new Buffer([
    SMP_PAIRING_REQUEST,
    0x03, // IO capability: NoInputNoOutput
    0x00, // OOB data: Authentication data not present
    0x01, // Authentication requirement: Bonding - No MITM
    0x10, // Max encryption key size
    0x05, // Initiator key distribution
    0x07  // Responder key distribution
  ]);
 
  this.write(this._preq);
};
```
I also added some hexdumping so I could see what data was getting passed back and forth. The debug output:
```
  hci create le conn - writing: 010d20196000300000019043f96676600006000c000000c80004000600 +1ms
  hci onSocketData: 040f0400010d20 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 15 +0ms
  hci 		status = 0 +0ms
  hci 		cmd = 8205 +0ms
  hci onSocketData: 043e130100400000019043f96676600c000000c80005 +1s
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 1 +0ms
  hci 		LE meta event status = 0 +0ms
  hci 		LE meta event data = 400000019043f96676600c000000c80005 +0ms
  hci 			handle = 64 +1ms
  hci 			role = 0 +0ms
  hci 			address type = random +0ms
  hci 			address = 60:76:66:f9:43:90 +0ms
  hci 			interval = 15 +0ms
  hci 			latency = 0 +0ms
  hci 			supervision timeout = 2000 +0ms
  hci 			master clock accuracy = 5 +0ms
  att 60:76:66:f9:43:90: write: 020001 +2ms
  hci write acl data pkt - writing: 024000070003000400020001 +0ms
  hci onSocketData: 040f0400011620 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 15 +0ms
  hci 		status = 0 +0ms
  hci 		cmd = 8214 +0ms
  hci onSocketData: 040e04010c2000 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 0 +0ms
[... 256 lines cut ...]
      Initiator Key Distribution:  LTK  
      Responder Key Distribution:  LTK IRK CSRK
2016-02-22 15:25:03.330587 < ACL data: handle 64 flags 0x00 dlen 21
    SMP: Pairing Confirm (0x03)
      key 149a78aaf3028cf0dc73546e971f56dc
2016-02-22 15:25:03.331429 < ACL data: handle 64 flags 0x00 dlen 21
    SMP: Pairing Confirm (0x03)
      key cb70e792e5d738f3764cf335e6f405ce
2016-02-22 15:25:03.344194 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2016-02-22 15:25:03.345202 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2016-02-22 15:25:03.478438 > ACL data: handle 64 flags 0x02 dlen 6
    SMP: Pairing Failed (0x05)
      reason 0x08
      Reason Unspecified Reason
2016-02-22 15:25:03.478477 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x05
    Reason: Authentication Failure
2016-02-22 15:25:03.479191 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2016-02-22 15:25:03.494194 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x16
    Reason: Connection Terminated by Local Host
```

sandeepmistry, 2016-02-27:
@dethredic this is making more sense, as you noticed noble currently does not support incoming SMP Pairing Request's. All of the peripheral's I have access to do not initiate pairing when a secure characteristic is accessed. They return a ``ATT_ECODE_AUTHENTICATION`` or ``ATT_ECODE_AUTHORIZATION`` error code, then noble kicks off the pairing process.

This matches what's going on it @deadprogram's debug log as well (but not in the HCI dump provided earlier for whatever reason):

```
  hci   event type = 2 +0ms
  hci           cid = 6 +0ms
  hci           handle = 3585 +0ms
  hci           data = 0b01 +0ms
```

With your changes, pairing is failing:
```
2016-02-22 15:25:03.478438 > ACL data: handle 64 flags 0x02 dlen 6
    SMP: Pairing Failed (0x05)
      reason 0x08
      Reason Unspecified Reason
```

I suggest you take a look at the bleno SMP stack, there are some workarounds that are needed because the kernel was interfering, and kernel HCI mgmt sockets are also needed to load the key:
* https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/smp.js

Another thing to check is what the kernel is doing with noble's SMP stack is disabled. If it does all the work, noble can just ignore SMP data in this scenario.

BasKiers, 2016-11-07:
I also experienced this issue. Although I do not have a hcidump I think we've already established that this exception is caused because of incoming SMP Pairing Requests which are not supported. I've created a pull request with a temporary solution (as long as incoming requests are not supported) which only registers the onAclStreamDataBinded function after sendPairingRequest is called and therefore this._preq is set.

For what it is worth, my debug log of this crash:
```
Mon, 07 Nov 2016 13:48:28 GMT noble scanStop
Mon, 07 Nov 2016 13:48:28 GMT hci onSocketData: 043e22020100002cfe0cbf7a8016020105030300cb0e0948544320425320324333343430b2
Mon, 07 Nov 2016 13:48:28 GMT hci 	event type = 4
Mon, 07 Nov 2016 13:48:28 GMT hci 	sub event type = 62
Mon, 07 Nov 2016 13:48:28 GMT hci 		LE meta event type = 2
Mon, 07 Nov 2016 13:48:28 GMT hci 		LE meta event status = 1
Mon, 07 Nov 2016 13:48:28 GMT hci 		LE meta event data = 00002cfe0cbf7a8016020105030300cb0e0948544320425320324333343430b2
Mon, 07 Nov 2016 13:48:28 GMT hci 			type = 0
Mon, 07 Nov 2016 13:48:28 GMT hci 			address = 80:7a:bf:0c:fe:2c
Mon, 07 Nov 2016 13:48:28 GMT hci 			address type = public
Mon, 07 Nov 2016 13:48:28 GMT hci 			eir = 020105030300cb0e0948544320425320324333343430
Mon, 07 Nov 2016 13:48:28 GMT hci 			rssi = -78
Mon, 07 Nov 2016 13:48:28 GMT gap advertisement = {"localName":"HTC BS 2C3440","serviceData":[],"serviceUuids":["cb00"],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]}
Mon, 07 Nov 2016 13:48:28 GMT noble emitted discover
Mon, 07 Nov 2016 13:48:28 GMT hci onSocketData: 010c20020000
Mon, 07 Nov 2016 13:48:28 GMT hci 	event type = 1
Mon, 07 Nov 2016 13:48:28 GMT hci 		cmd = 8204
Mon, 07 Nov 2016 13:48:28 GMT hci 		data len = 2
Mon, 07 Nov 2016 13:48:28 GMT hci 			LE enable scan command
Mon, 07 Nov 2016 13:48:28 GMT hci 			enable scanning = false
Mon, 07 Nov 2016 13:48:28 GMT hci 			filter duplicates = false
Mon, 07 Nov 2016 13:48:28 GMT hci disconnect - writing: 01060403400013
Mon, 07 Nov 2016 13:48:28 GMT hci onSocketData: 040e04010c2000
Mon, 07 Nov 2016 13:48:28 GMT hci 	event type = 4
Mon, 07 Nov 2016 13:48:28 GMT hci 	sub event type = 14
Mon, 07 Nov 2016 13:48:28 GMT hci 		cmd = 8204
Mon, 07 Nov 2016 13:48:28 GMT hci 		status = 0
Mon, 07 Nov 2016 13:48:28 GMT hci 		result =
Mon, 07 Nov 2016 13:48:28 GMT hci onSocketData: 010d20196000600000002cfe0cbf7a800006000c000000c80000000000
Mon, 07 Nov 2016 13:48:28 GMT hci 	event type = 1
Mon, 07 Nov 2016 13:48:28 GMT hci 		cmd = 8205
Mon, 07 Nov 2016 13:48:28 GMT hci 		data len = 25
Mon, 07 Nov 2016 13:48:28 GMT hci onSocketData: 040f0400010d20
Mon, 07 Nov 2016 13:48:28 GMT hci 	event type = 4
Mon, 07 Nov 2016 13:48:28 GMT hci 	sub event type = 15
Mon, 07 Nov 2016 13:48:28 GMT hci 		status = 0
Mon, 07 Nov 2016 13:48:28 GMT hci 		cmd = 8205
Mon, 07 Nov 2016 13:48:28 GMT hci onSocketData: 043e130100410000002cfe0cbf7a800c000000c80005
Mon, 07 Nov 2016 13:48:28 GMT hci 	event type = 4
Mon, 07 Nov 2016 13:48:28 GMT hci 	sub event type = 62
Mon, 07 Nov 2016 13:48:28 GMT hci 		LE meta event type = 1
Mon, 07 Nov 2016 13:48:28 GMT hci 		LE meta event status = 0
Mon, 07 Nov 2016 13:48:28 GMT hci 		LE meta event data = 410000002cfe0cbf7a800c000000c80005
Mon, 07 Nov 2016 13:48:28 GMT hci 			handle = 65
Mon, 07 Nov 2016 13:48:28 GMT hci 			role = 0
Mon, 07 Nov 2016 13:48:28 GMT hci 			address type = public
Mon, 07 Nov 2016 13:48:28 GMT hci 			address = 80:7a:bf:0c:fe:2c
Mon, 07 Nov 2016 13:48:28 GMT hci 			interval = 15
Mon, 07 Nov 2016 13:48:28 GMT hci 			latency = 0
Mon, 07 Nov 2016 13:48:28 GMT hci 			supervision timeout = 2000
Mon, 07 Nov 2016 13:48:28 GMT hci 			master clock accuracy = 5
@@@@@@ LE CONN COMPLETE 807abf0cfe2c
Mon, 07 Nov 2016 13:48:28 GMT att 80:7a:bf:0c:fe:2c: write: 020001
Mon, 07 Nov 2016 13:48:28 GMT hci write acl data pkt - writing: 024100070003000400020001
unlock scanning 1
Mon, 07 Nov 2016 13:48:28 GMT hci onSocketData: 011620024100
Mon, 07 Nov 2016 13:48:28 GMT hci 	event type = 1
Mon, 07 Nov 2016 13:48:28 GMT hci 		cmd = 8214
Mon, 07 Nov 2016 13:48:28 GMT hci 		data len = 2
Mon, 07 Nov 2016 13:48:28 GMT hci onSocketData: 040f0400011620
Mon, 07 Nov 2016 13:48:28 GMT hci 	event type = 4
Mon, 07 Nov 2016 13:48:28 GMT hci 	sub event type = 15
Mon, 07 Nov 2016 13:48:28 GMT hci 		status = 0
Mon, 07 Nov 2016 13:48:28 GMT hci 		cmd = 8214
Mon, 07 Nov 2016 13:48:28 GMT hci onSocketData: 043e0c040041000100000000000000
Mon, 07 Nov 2016 13:48:28 GMT hci 	event type = 4
[... 13 lines cut ...]
Mon, 07 Nov 2016 13:48:28 GMT hci 		cid = 6
Mon, 07 Nov 2016 13:48:28 GMT hci 		handle = 65
Mon, 07 Nov 2016 13:48:28 GMT hci 		data = 02030001100507
Mon, 07 Nov 2016 13:48:28 GMT hci set scan enabled - writing: 010c20020001
Mon, 07 Nov 2016 13:48:28 GMT hci disconnect - writing: 01060403400013
Mon, 07 Nov 2016 13:48:28 GMT hci disconnect - writing: 01060403410013
buffer.js:219
      length += list[i].length;
                       ^

TypeError: Cannot read property 'length' of undefined
    at Function.Buffer.concat (buffer.js:219:24)
    at Object.c1 (/opt/homey-client/node_modules/noble/lib/hci-socket/crypto.js:8:19)
    at Smp.handlePairingResponse (/opt/homey-client/node_modules/noble/lib/hci-socket/smp.js:86:12)
    at Smp.onAclStreamData (/opt/homey-client/node_modules/noble/lib/hci-socket/smp.js:57:10)
    at emitTwo (events.js:92:20)
    at emit (events.js:172:7)
    at AclStream.push (/opt/homey-client/node_modules/noble/lib/hci-socket/acl-stream.js:35:10)
    at NobleBindings.onAclDataPkt (/opt/homey-client/node_modules/noble/lib/hci-socket/bindings.js:277:15)
    at emitThree (events.js:97:13)
    at emit (events.js:175:7)
    at Hci.onSocketData (/opt/homey-client/node_modules/noble/lib/hci-socket/hci.js:473:14)
    at emitOne (events.js:77:13)
    at BluetoothHciSocket.emit (events.js:169:7)
```

maxdenisov, 2017-01-26:
Wiced sense is working with fix from https://github.com/sandeepmistry/noble/pull/495

WaeCo, 2018-10-25:
Any progress or at least a plan how to solve this?

#495 was also working for me. Is there a reason why it was not merged?