noble/noble #296: Peripheral disconnects after 1 minute - only with Noble

PimButton, 2016-01-04:
Hi Sandeep! 
Unfortunately I am still experiencing the problem I described in #269 and #268. 

## Setup ##
Windows 7, 32bit, most recent version of Noble. 
Custom peripheral, based on NRF81422. It broadcasts a service and a characteristic with an orientation reading from an accelerometer attached to the NRF. 
With Noble I connect to the peripheral and set a notify on the accelerometer characteristic. Whenever the device is turned, Noble receives an update and displays the correct value. 

## Problem ##
After 1 minute (+- 5 seconds) the connection is disconnected and Noble fires the ```peripheral.on('disconnect', callback);``` Debugging the NRF tells me the NRF initiates the disconnect. So far I have not been able to find out why it disconnects. 

## Analysis ##
With an iPad application like [BLE Tool](https://itunes.apple.com/nl/app/bluetooth-smart-tool/id525235616?mt=8) or Apple's Bluetooth Explorer the disconnect does not happen. Therefore I suspect that something is different in the way the connection between my peripheral and Noble is handled, and that that causes my peripheral to disconnect. Unfortunately I do not have a deep enough understanding of the BLE protocol to find those differences myself. 
I used Wireshark 1.12.8 to log the communication twice: 
1. my peripheral & noble 
2. my peripheral & BLE Tool (iPad application). 
Had to use version 1, otherwise the BLE logger would not work. @sandeepmistry maybe you can try version 1 on the Mac to open the files? I hope they will open fine then. I'll install wireshark v1 now on my mac and will let you know if it works!

I also tried @george-hawkins remark to enable Debug for the att file (using ```set debug=att```) but after the connection is established I do not receive any more debug output, even after the connection is disconnected. I think this proves that the error is within a different part of the Noble code? 

Hope we can find a solution!

sandeepmistry, 2016-01-05:
@PimButton thanks for opening a new issue.

Just to confirm, you've tried out with #296? Since this has been merged, you can tryout via ```npm install sandeepmistry/noble```.

Here's what I see in Wireshark when I try to open your packet captures from https://github.com/sandeepmistry/noble/issues/269#issuecomment-166946562. This is after I set the DLT=157 payload in preferences:

[image]

[image]

sandeepmistry, 2016-01-05:
@PimButton just saw your edit (I was reading the email Github set previously). I don't have access to a Windows machine off hand at the moment, but could look into using a VM.

PimButton, 2016-01-05:
@sandeepmistry thanks for your quick reply!
Yes, I have checked with a new, fresh install after the #296 merge but that did not resolve the issue unfortunately. I don't even see any debug output (as mentioned above) so I think it must be in some other part of the code. 

However, I have managed to export to Wireshark file to a quite readable HTML version. It's not great, but it should do the trick. Will probably save you a lot of time messing around with a VM. 
NRF to Noble can be found here: http://pimknops.com/klanten/2015/Debug-session-html-version/NRF-to-Noble.xml
and NRF to iPad is here: http://pimknops.com/klanten/2015/Debug-session-html-version/NRF-to-iPad.xml
Just for reference, this is my wireshark screen for the first:
[image]
and the second:
[image]
The whole package is available for download here: http://pimknops.com/klanten/2015/NRF51822-Noble-debug-session.zip

Hope this helps!

PimButton, 2016-01-05:
Ok, with the help of the guys over at Nordic Semi I have resolved the issue! 
It turns out that the firmware on the NRF sends out an L2Cap connection parameter update to the master (Noble in this case) when the connection is established. If the NRF does not receive a response the L2Cap connection parameter update times out and the connection is disconnected. 

With their help it was very easy to disable the L2Cap connection parameter update altogether, so the timeout doesn't happen anymore and the connection stays alive. 

@sandeepmistry I hope this has not taken up too much of your time, it turns out it had not that much to do with Noble after all. It might be an idea though to at least respond to these kinds of update requests in Noble (maybe just acknowledge the settings already in use?) since I think this is default behavior of most of the NRF51822's out there, at least with custom firmware.

sandeepmistry, 2016-01-08:
@PimButton great find!

Can you be more specific on the code change you had to make to your nRF51822 firmware? Then I can reproduce and fix. This issue might be specific to noble + Windows, and Linux might be handling this case for noble.

sandeepmistry, 2016-01-23:
@PimButton ping ...

sandeepmistry, 2016-01-30:
Closing due to lack of activity, feel free to re-open with details requested from https://github.com/sandeepmistry/noble/issues/296#issuecomment-169847449.

closed 2016-01-30 by sandeepmistry

PimButton, 2016-03-13:
I'm sorry, completed missed that I had not answered here. 
The code change was very simple: not calling ```conn_params_init()``` in the main loop of the NRF51822 firmware. The case is discussed by one of Nordic's engineers here: https://devzone.nordicsemi.com/question/63233/nrf51822-disconnects-connection-after-1-min/?answer=63259#post-id-63259