noble/noble #299: process hanging?

AKST, 2016-01-09:
Is there any reason why nobel would cause a program to hang, even after closing `noble.stopScanning()` has been called and no peripherals are connected?

sandeepmistry, 2016-01-09:
@AKST noble continuously monitoring for adapter state change events: https://github.com/sandeepmistry/noble#adapter-state-change

AKST, 2016-01-13:
is there a means to stop this once I no longer wish to connect any BLE devices?

sandeepmistry, 2016-01-17:
@AKST not currently. What is the goal for stopping this?

Is using ```process.exit(0)``` to exit you app an option?

AKST, 2016-01-18:
Well I’m using this library as part of another library, and I don’t think it’s fair to expect the users of my own library to have to call process.exit(0) to close there applications which is why I’m seeking a means to stop this once the library is no longer being used.

sandeepmistry, 2016-01-23:
@AKST I've renamed this issue, and labelled it as an enhancement request.

AKST, 2016-01-23:
Thanks @sandeepmistry, that's fair enough.

I would be happy to help out where I can to progress this issue, so if you feel like delegating any work just let me know, as I went poking through quite a bit of the source to try and find a solution myself before posting this.

Otherwise I'll be more than happy to just test any changes.

sandeepmistry, 2016-01-24:
If you could help out that would be great!

I think we'll have to add equivalent API's in noble's dependencies, as they continuously listen:
 * https://github.com/sandeepmistry/node-xpc-connection - for OS X
 * https://github.com/sandeepmistry/node-bluetooth-hci-socket - for non-OS X

AKST, 2016-01-27:
Awesome, I start having a look around the weekend!

— Angus

recursify, 2016-01-27:
@sandeepmistry Can we also add an  API to *start* noble?

For example, this simple script will hang:

```
require('noble')
```

IMO - requiring a module *should not* hang.  So something like:

```
var noble = require('noble');
noble.start();
// Do stuff ..
noble.stop();
```

sandeepmistry, 2016-02-07:
@recursify this would be a breaking change, something to start discussing in #325.

sandeepmistry, 2016-06-06:
@AKST any progress on this?

AKST, 2016-06-06:
Eh unfortunately not, I probably shouldn't have put my name forward to eagerly. Feel free to delegate it someone else

jayalfredprufrock, 2017-02-21:
aww, too bad this never made it into a release. I'm in the same boat as OP. Another argument for this feature is that its nice to be able to fully shut down modules when using test libraries that hang until all processes exit.

I think it does make sense to also add a corresponding `start()` method. To avoid a breaking change, we could call `start()` on initialization by default and have an option to disable that behavior.

I'll look into picking this up in a month or two if I can find some extra time but I'd be forever grateful if somebody already familiar with the internals could knock this out!

oyooyo, 2017-03-10:
I would highly welcome introducing a `stop()` method as well. As I understand, process.exit isn't the recommended way of stopping a node program.

I'm in favor of adding a `start()` method as well. When it comes to the problem of this being a breaking change:
Wouldn't it be possible to auto-start noble if a function is called that requires noble to be started, but it hasn't been started yet? For example by adding a little check in noble.on('stateChange') or noble.startScanning()?

sandeepmistry, 2017-03-13:
> I would highly welcome introducing a stop() method as well. As I understand, process.exit isn't the recommended way of stopping a node program.

@oyooyo please open a pull request to get this going

oyooyo, 2017-03-20:
@sandeepmistry I opened a pull request as you requested: #577 
The "solution" I provided probably doesn't really work properly, but might eventually serve as a kind of starting point for a proper stop() function.

readeral, 2018-08-26:
Are these pull requests going to be considered?

RobertMcReed, 2019-01-10:
I'm a little late to the party, but would it suit your needs to run your noble dependency in a child process?

Although it's not the most convenient work around, it does work. 

There are a bunch of different ways you could implement this (and more you could do with it), but here's an example where you can start and stop noble without messing with the main process. 

You can pass messages from the child to the main process, handle errors from the child process, and if the main process exits so will the child.

`nobleController.js`: The parent process, import where you want to "start" and "stop" noble.
```
const { join } = require('path');
const { fork } = require('child_process');

const startNoble = ({ handleMessage, handleClose, handleError }) => {
  // start the child process
  const nobleProcess = fork(join(__dirname, 'handleNoble.js'));
  
  // handle data sent from the child process
  nobleProcess.on('message', handleMessage);

  // in the event that the you need to handle noble exiting
  nobleProcess.on('close', (exitCode) => {
    // non-zero exit code if an error occurs
    exitCode ? handleError(exitCode) : handleClose();
  });

  // bind a function for terminating the child process
  const stop = nobleProcess.kill.bind(nobleProcess);

  // ensure noble exits if the parent exits first
  process.on('exit', stop);

  // return the stop function to terminate noble from the main process 
  return stop;
};

module.exports = startNoble;
```

`handleNoble.js`: The child process, where you write all of your actual noble code.
```
const noble = require('noble');

// main logic for handling noble code
const onDiscovery = (peripheral) => {
  const {
    address: mac,
    advertisement: { localName: name }
  } = peripheral;

  if (!name) return;

  // send data to the parent process
  process.send({ mac, name });

  if (Math.random() > 0.7) {
    console.log('__NOBLE_SELF_DESTRUCT__\n');

    // terminate noble without exiting the main process
    process.exit();
  };
  
  if (Math.random() > 0.8) {
    // this will throw an error for sure
    brokenCode();
  };
}

// console.logs will appear in the main window
console.log('Scanning for devices...\n');
noble.startScanning([], true);
noble.on('discover', onDiscovery);
```

`index.js`: Where you might use your "wrapped" noble module.
```
const startNoble = require('./nobleController');

let running = true;

const handleMessage = ({ mac, name }) => {
  console.log(`Mac Address: ${mac} --- Local Name: ${name}\n`);
};

const handleClose = () => {
  running = false;
  console.log('It looks like noble has terminated cleanly.\n');
};

const handleError = (exitCode) => {
  console.log(`\nIt looks like something broke in the child process, exiting with code ${exitCode}.\n`);
  process.exit(exitCode);
};

// start noble and capture the stop function
const stop = startNoble({ handleMessage, handleClose, handleError });

// optionally stop the child process
setTimeout(() => {
  if (running) {
    console.log('Terminating noble from main process.\n');
    stop();
  }
}, 5000);

// the main process stays alive even after the noble process has terminated
setTimeout(() => {
  console.log('Running more code after noble has exited...');
}, 10000);
```

I'd be curious as to whether others think this is an ok solution, and what the potential downsides of this approach may be.

Cheers

jhfoo, 2019-02-10:
> I'm a little late to the party, but would it suit your needs to run your noble dependency in a child process?
>
> Although it's not the most convenient work around, it does work.
>
> There are a bunch of [...]

Not that it's a bad idea; just calling out that it may introduce a lot of traffic through IPC.