noble/noble #300: Way to get callback if some other process changes scan parameters?

bradjc, 2016-01-11:
I'm running into an issue where I'm trying to have a long lived noble script scan continuously for BLE advertisements, but if anything else on the system uses the BLE/HCI subsystem it can change the scanning parameters unbeknownst to the noble script. For example, running

    sudo hcitool lescan

while the noble script is running causes the noble script to only receive unique advertisements. Ending `lescan` stops all advertisements.

I think the best way to address this is to augment `Gap.prototype.onHciLeScanEnableSet` ([here](https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/gap.js#L40)) with an `else` statment that emits a callback if the scan parameters changed outside of the script. Then the script can change them back.

Ideally this wouldn't be an issue, but it's hard to make a robust application if there is no way to detect conflicts like these.

sandeepmistry, 2016-01-17:
noble might be able to listen for HCI commands (filter needs to be changed), and react based on the status.

* Here is where the filter is currently set: https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/hci.js#L117
* Here is where socket data is processed: https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/hci.js#L371

> I think the best way to address this is to augment Gap.prototype.onHciLeScanEnableSet (here) with an else statment that emits a callback if the scan parameters changed outside of the script. Then the script can change them back.

That works for me.

sandeepmistry, 2016-02-21:
Closed via #301.

closed 2016-02-21 by sandeepmistry