noble/noble #302: Install error on Intel Edison

prachaquant, 2016-01-20:
Hi,
I'm using Yocto 2.1 and node 0.12.2
Trying to install noble.
```
 root@EDI-X:/node_app_slot# npm install noble                                                                                                                                                               
npm WARN package.json LightWave@0.0.1 No repository field.                                                                                                                                                                  
npm WARN optional dep failed, continuing xpc-connection@0.1.4                                                                                                                                                               
\                                                                                                                                                                                                                           
> usb@1.1.1 install /home/root/.node_app_slot/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb                                                                                                         
> node-pre-gyp install --fallback-to-build                                                                                                                                                                                  
                                                                                                                                                                                                                            
gyp WARN install got an error, rolling back install                                                                                                                                                                         
gyp ERR! configure error                                                                                                                                                                                                    
gyp ERR! stack Error: socket hang up                                                                                                                                                                                        
gyp ERR! stack     at createHangUpError (_http_client.js:215:15)                                                                                                                                                            
gyp ERR! stack     at Socket.socketOnEnd (_http_client.js:300:23)                                                                                                                                                           
gyp ERR! stack     at Socket.emit (events.js:129:20)                                                                                                                                                                        
gyp ERR! stack     at _stream_readable.js:908:16                                                                                                                                                                            
gyp ERR! stack     at process._tickCallback (node.js:355:11)                                                                                                                                                                
gyp ERR! System Linux 3.10.17-poky-edison+                                                                                                                                                                                  
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "--fallback-to-build" "--module=/home/root/.node_app_slot/node_modules/noble/node_modules/bluetoo
th-hci-socket/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/home/root/.node_app_slot/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding"      
gyp ERR! cwd /home/root/.node_app_slot/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb                                                                                                                
gyp ERR! node -v v0.12.2                                                                                                                                                                                                    
gyp ERR! node-gyp -v v1.0.3                                                                                                                                                                                                 
gyp ERR! not ok                                                                                                                                                                                                             
node-pre-gyp ERR! build error                                                                                                                                                                                               
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/home/root/.node_app_slot/node_modules/no
ble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/root/.node_app_slot/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/us
b/src/binding' (1)                                                                                                                                                                                                          
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/root/.node_app_slot/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/node_modules/node-pre-gyp/lib/util/compile.js:83:29)               
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:110:17)                                                                                                                                                         
node-pre-gyp ERR! stack     at maybeClose (child_process.js:1015:16)                                                                                                                                                        
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1087:5)                                                                                                                                
node-pre-gyp ERR! System Linux 3.10.17-poky-edison+                                                                                                                                                                         
node-pre-gyp ERR! command "node" "/home/root/.node_app_slot/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"                           
node-pre-gyp ERR! cwd /home/root/.node_app_slot/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb                                                                                                       
node-pre-gyp ERR! node -v v0.12.2                                                                                                                                                                                           
node-pre-gyp ERR! node-pre-gyp -v v0.6.4                                                                                                                                                                                    
node-pre-gyp ERR! not ok                                                                                                                                                                                                    
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/home/root/.node_app_slot/node_modules/noble/node_modules/bluetooth-hci-
socket/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/root/.node_app_slot/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding' (1)             
[... cut ...]
npm WARN optional dep failed, continuing bluetooth-hci-socket@0.4.2                                                                                                                                                         
noble@1.3.0 node_modules/noble                                                                                                                                                                                              
├── bplist-parser@0.0.6                                                                                                                                                                                                     
└── debug@2.2.0 (ms@0.7.1)   
```

deadprogram, 2016-01-20:
Hi, @prachaquant did to happen to install the dependencies on the Edison? Check out https://github.com/w4ilun/edison-guides/wiki/Configure-Intel-Edison-for-BLE---Bluetooth-Smart-Development

prachaquant, 2016-01-20:
I followed a similar guide.

Can I use this version?                                                                                                                                                                       
Package bluez5-dev (5.24-r0) installed in root is up to date.

I have a problem with opkg update
Collected errors:                                                                                                                                                                                                           
 * opkg_download: Failed to download http://repo.opkg.net/edison/repo/all/Packages.gz, wget returned 4.                                                                                                                     
 * opkg_download: Failed to download http://repo.opkg.net/edison/repo/core2-32/Packages.gz, wget returned 4.

deadprogram, 2016-01-20:
Is your Edison itself able to access the Internet? You need that for the npm install as well.

That version of bluez looks fine to me.

prachaquant, 2016-01-20:
It can ping to google and use npm install only have problem on opkg update.
```root@EDI-X:~# ping google.com                                                                                                                                                                                               
PING google.com (216.58.196.206): 56 data bytes                                                                                                                                                                             
64 bytes from 216.58.196.206: seq=0 ttl=52 time=85.137 ms                                                                                                                                                                   
64 bytes from 216.58.196.206: seq=1 ttl=52 time=74.504 ms                                                                                                                                                                   
64 bytes from 216.58.196.206: seq=2 ttl=52 time=48.183 ms                                                                                                                                                                   
64 bytes from 216.58.196.206: seq=3 ttl=52 time=73.369 ms                                                                                                                                                                   
64 bytes from 216.58.196.206: seq=4 ttl=52 time=82.831 ms                                                                                                                                                                   
64 bytes from 216.58.196.206: seq=5 ttl=52 time=82.341 ms    ```

deadprogram, 2016-01-20:
Perhaps you need to run `opkg install libusb-1.0-dev` as well?

deadprogram, 2016-01-20:
Those errors from the npm install certainly look like a server that is unreachable.

prachaquant, 2016-01-20:
There are problem with my isp. That new device using http protocol will redirect to their landing page. But I fix now. Thank you.

closed 2016-01-20 by prachaquant