noble/noble #303: How to Speed UP Noble?

boredSinan, 2016-01-20:
Hi, 

Can I speed up the Noble project with editing some parameters in the lib or any other way?

Here you can see the current output of my Noble-based application:

Date | RSSI | Beacon_ID | Distance_in_m
------------ | ------------ | ------------- | ------------
2016-01-20 09:23:25| -53| 410| 0.44668359215096315
2016-01-20 09:23:26| -47| 410| 0.22387211385683395
2016-01-20 09:23:26| -80| 450| 3.548133892335755
2016-01-20 09:23:27| -48| 410| 0.251188643150958
2016-01-20 09:23:27| -78| 450| 2.8183829312644537
2016-01-20 09:23:28| -47| 410| 0.22387211385683395
2016-01-20 09:23:28| -81| 450| 3.9810717055349722
2016-01-20 09:23:28| -50| 411| 0.31622776601683794
2016-01-20 09:23:29| -48| 410| 0.251188643150958
2016-01-20 09:23:29| -52| 449| 0.14125375446227545
2016-01-20 09:23:30| -77| 450| 2.51188643150958
2016-01-20 09:23:30| -54| 449| 0.1778279410038923
2016-01-20 09:23:31| -47| 410| 0.22387211385683395
2016-01-20 09:23:31| -75| 450| 1.9952623149688795
2016-01-20 09:23:31| -53| 449| 0.15848931924611134
2016-01-20 09:23:31| -57| 411| 0.7079457843841379

Time in seconds | Detected Beacons
------------ | ------------ |
09:23:25 | 1
09:23:26| 2
09:23:27| 2
09:23:28| 3
09:23:29| 2
09:23:30| 2
09:23:31| 4

So at e.g. "09:23:31" there are 4 detected beacons and e.g. at "09:23:25" there is only 1 beacon detected. Of course this is random, but I would like to speed up Noble for detecting at least 5 beacons within every second.

sandeepmistry, 2016-01-23:
@boredSinan this might not be related to noble at all, what advertisement interval are your beacons configured at? Have you tried using something other than noble and got different results?

noble current uses a scan interval of 10ms: https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/hci.js#L239-L240

boredSinan, 2016-01-25:
Hi,
Till yet I didn't try any other project than Noble.

In the following results of different parameter-values for the "writeUInt16LE()" function are listed:

a)
```javascript
cmd.writeUInt16LE(0x0001, 5);
cmd.writeUInt16LE(0x0001, 7);
```
Date | RSSI | Beacon_ID | Distance_in_m
------------ | ------------ | ------------- | ------------
2016-01-25 08:57:50| -63| 450| 0.5011872336272722
2016-01-25 08:57:51| -59| 449| 0.31622776601683794
2016-01-25 08:57:52| -63| 449| 0.5011872336272722
2016-01-25 08:57:52| -64| 450| 0.5623413251903491
2016-01-25 08:57:53| -57| 449| 0.251188643150958
2016-01-25 08:57:53| -63| 450| 0.5011872336272722
2016-01-25 08:57:54| -59| 449| 0.31622776601683794
2016-01-25 08:57:54| -63| 450| 0.5011872336272722
2016-01-25 08:57:55| -63| 449| 0.5011872336272722
2016-01-25 08:57:55| -63| 450| 0.5011872336272722
2016-01-25 08:57:56| -57| 449| 0.251188643150958
2016-01-25 08:57:56| -65| 450| 0.6309573444801932
2016-01-25 08:57:57| -62| 449| 0.44668359215096315
2016-01-25 08:57:57| -63| 450| 0.5011872336272722

b)
```javascript
cmd.writeUInt16LE(0x0010, 5);
cmd.writeUInt16LE(0x0010, 7);
```
Date | RSSI | Beacon_ID | Distance_in_m
------------ | ------------ | ------------- | ------------
2016-01-25 09:04:05| -61| 449| 0.39810717055349726
2016-01-25 09:04:05| -67| 450| 0.7943282347242815
2016-01-25 09:04:05| -53| 412| 0.44668359215096315
2016-01-25 09:04:05| -53| 412| 0.44668359215096315
2016-01-25 09:04:05| -60| 411| 1
2016-01-25 09:04:05| -53| 410| 0.44668359215096315
2016-01-25 09:04:06| -58| 449| 0.28183829312644537
2016-01-25 09:04:06| -66| 450| 0.7079457843841379
2016-01-25 09:04:06| -59| 412| 0.8912509381337456
2016-01-25 09:04:06| -61| 411| 1.1220184543019633
2016-01-25 09:04:06| -54| 410| 0.5011872336272722
2016-01-25 09:04:07| -61| 449| 0.39810717055349726
2016-01-25 09:04:07| -67| 450| 0.7943282347242815
2016-01-25 09:04:07| -53| 412| 0.44668359215096315
2016-01-25 09:04:07| -60| 411| 1
2016-01-25 09:04:07| -59| 410| 0.8912509381337456

c)
```javascript
cmd.writeUInt16LE(0x0100, 5);
cmd.writeUInt16LE(0x0100, 7);
```
Date | RSSI | Beacon_ID | Distance_in_m
------------ | ------------ | ------------- | ------------
2016-01-25 09:05:16| -60| 410| 0.5011872336272722
2016-01-25 09:05:16| -54| 410| 0.5011872336272722
2016-01-25 09:05:16| -58| 449| 0.28183829312644537
2016-01-25 09:05:16| -65| 450| 0.6309573444801932
2016-01-25 09:05:16| -56| 412| 0.6309573444801932
2016-01-25 09:05:17| -60| 411| 1
2016-01-25 09:05:17| -58| 449| 0.28183829312644537
2016-01-25 09:05:17| -64| 450| 0.5623413251903491
2016-01-25 09:05:17| -63| 412| 1.4125375446227544
2016-01-25 09:05:18| -60| 411| 1
2016-01-25 09:05:18| -58| 449| 0.28183829312644537
2016-01-25 09:05:18| -66| 450| 0.7079457843841379
2016-01-25 09:05:18| -62| 412| 1.2589254117941673
2016-01-25 09:05:19| -53| 410| 0.44668359215096315
2016-01-25 09:05:19| -60| 449| 0.35481338923357547
2016-01-25 09:05:19| -66| 450| 0.7079457843841379
2016-01-25 09:05:19| -62| 412| 1.2589254117941673

d)
```javascript
cmd.writeUInt16LE(0x1000, 5);
cmd.writeUInt16LE(0x1000, 7);
```
Date | RSSI | Beacon_ID | Distance_in_m
------------ | ------------ | ------------- | ------------
2016-01-25 09:08:52| -59| 410| 0.8912509381337456
2016-01-25 09:08:52| -52| 411| 0.39810717055349726
2016-01-25 09:08:52| -57| 449| 0.251188643150958
2016-01-25 09:08:52| -62| 412| 1.2589254117941673
2016-01-25 09:08:53| -62| 410| 1.2589254117941673
2016-01-25 09:08:53| -61| 411| 1.1220184543019633
2016-01-25 09:08:53| -56| 449| 0.22387211385683395
2016-01-25 09:08:53| -79| 450| 3.1622776601683795
2016-01-25 09:08:53| -62| 412| 1.2589254117941673
2016-01-25 09:08:54| -63| 411| 1.2589254117941673
2016-01-25 09:08:54| -62| 411| 1.2589254117941673
2016-01-25 09:08:54| -55| 449| 0.19952623149688797
2016-01-25 09:08:54| -77| 450| 2.51188643150958
2016-01-25 09:08:54| -60| 412| 1
2016-01-25 09:08:55| -61| 411| 1.1220184543019633
2016-01-25 09:08:55| -61| 411| 1.1220184543019633
2016-01-25 09:08:55| -58| 449| 0.28183829312644537
2016-01-25 09:08:55| -61| 450| 0.39810717055349726
2016-01-25 09:08:55| -56| 412| 0.6309573444801932

All 5 beacons have been configured as follows:
Transmission-intervall: 1 s
Transmission-power: 0 dB

I noticed the following:
Increasing the parameter from "0x0001", "0x0010", "0x0100" to "0x1000" seems to increase noble scan interval.

Would you also agree? 
Are these test results acceptable and reasonable?

sandeepmistry, 2016-01-30:
> In cases a) and b) noble should detect more beacons in each second than in cases c) and d) but it doesn't! Why exactly not?

Do you have a summary table like in https://github.com/sandeepmistry/noble/issues/303#issue-127644642.

Have done tests with one or 2 beacons to see if they can be detected consistently? What about lowering the beacon's advertising interval.

> Are these test results acceptable and reasonable?

Hard for me to say now, lots of different factors ...

Have you had a chance to get more details on what the scan interval and window actually control?

boredSinan, 2016-02-01:
In the following there are the summary tables:

Time | Detected Beacons
------------ | ------------ |
09:23:25 | 1
09:23:26| 2

sandeepmistry, 2016-02-07:
Results b, c, and d look acceptable to me if you have 5 or 6 beacons in range.

closed 2016-02-07 by sandeepmistry

way2016, 2018-02-05:
As stated in "Bluetooth Core Specification v4.0 ,Volume 2, Part E, Section 7.8.12",
the unit of LE_Scan_Interval and LE_Scan_Window is 0.625 ms, that's why 0x10 = 16 x 0.625 = 10ms.

So the following setting is actually 0x100 x 0.625 = 160 ms.

But what's really important is:
"If both are set to the same value, scanning should run continuously."

> c) scan_interval = 100 ms
> 
> cmd.writeUInt16LE(0x0100, 5);
> cmd.writeUInt16LE(0x0100, 7);