noble/noble #304: startScanning with serviceUUIDs, linux issue

rosek86, 2016-01-22:
Hi, I have a device which sends advertisement with content as follows:

"serviceData":[{"uuid":"180a","data":{"type":"Buffer","data":[5]}}]

I'd like to detect devices which contains only '180a' UUID, so start scanning looks like this: 
noble.startScanning(['180a'], true, (err) => { ... });

devices are correctly detected on mac but on linux I have to remove serviceUUIDs do detect anything.

BR,
Krzysztof

sandeepmistry, 2016-01-23:
@rosek86 what peripheral are you scanning for? It doesn't look time your device is advertising the service, only including service data in the advertisement.

Could you please share the output of ```sudo hcidump -t -x``` while your app is running? Also, what does OS X show for the advertisement data?

rosek86, 2016-01-24:
@sandeepmistry it's property design based on Nordic chip, it sends service data in 0x16 data type [1]. Both iOS and noble on OS X detects that advertisements correctly, only noble on Linux doesn't.

Here is code snippet for iOS which I use to detect peripheral:

    let serviceUUIDs:[CBUUID] = [CBUUID(string: "180A")]
    centralManager.scanForPeripheralsWithServices(serviceUUIDs, options: options)

{"id":"09e90d452ea149e9950cdcc773506317","address":"dd:35:85:28:3d:3b","addressType":"unknown","connectable":true,"advertisement":{"localName":"TST","manufacturerData":{"type":"Buffer","data":[84,68,1,5,33,84,0,65,3,1,0,0,0,19,0,2,0,1]},"serviceData":[{"uuid":"180a","data":{"type":"Buffer","data":[7]}}],"serviceUuids":[]},"rssi":-56,"state":"disconnected"}

[1] https://www.bluetooth.org/en-us/specification/assigned-numbers/generic-access-profile

sandeepmistry, 2016-01-26:
@rosek86 I've push a change for this to master (https://github.com/sandeepmistry/noble/commit/4a324910d168467c85cd7ab1876c765facb14c70), you can test it out via ```npm install sandeepmistry/noble``` before it is released to npm.

Please let me know if there are any issues.

closed 2016-01-26 by sandeepmistry

rosek86, 2016-01-26:
@sandeepmistry I confirm that your fix works well.
Many thanks

sandeepmistry, 2016-01-30:
@rosek86 great! Thanks for verifying.