noble/noble #305: Scan response, differences between mac & linux

rosek86, 2016-01-22:
Hi,

It seems that on Linux Noble tries to merge advertisement and scan response together, which is a good way to handle this. However, if manufacturerData is presence in both; advertisement and scan response Noble returns only scan response field. Issue is with gap.js, line 137:  

https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/gap.js#L137

I believe that best way is to merge manufacturer data fields together, like it's done on iOS:

      case 0xff: // Manufacturer Specific Data
	if (advertisement.manufacturerData !== undefined) {
          advertisement.manufacturerData = Buffer.concat([advertisement.manufacturerData, bytes]);
  	} else {
          advertisement.manufacturerData = bytes;
        }

The mac version shows different behaviour, It seems that I get two separate events for advertisement and scan response.

BR,
Krzysztof

sandeepmistry, 2016-01-23:
@rosek86 can you please share an example of the manufacturer data in the advertisement data and scan data, and how CoreBluetooth on iOS and OS X report it?

rosek86, 2016-01-24:
@sandeepmistry This is something I get on OS X:
Advertisement:  <Buffer 54 44 01 d4 21 54 00 42 03 01 00 00 00 13 00 02 00 00>
Scan response: <Buffer 54 44 01 3b 3d 28 85 35 dd>

There are two separate events, first two bytes is manufacturer identifier which in this case is the same. The iOS merges advertisement and scan response manufacturer data into single field, removing second redundant manufacturer identifier.   

"kCBAdvDataManufacturerData": <544401d4 21540042 03010000 00130002 0000013b 3d288535 dd>]

Then on linux noble returns only scan response data:
<Buffer 54 44 01 3b 3d 28 85 35 dd>

I haven't checked what iOS does if manufacturer identifier is different in advertisement and scan response.

sandeepmistry, 2016-01-30:
@rosek86 thanks for sharing more details.

> I haven't checked what iOS does if manufacturer identifier is different in advertisement and scan response.

Could you please get more details this? Once we know the behaviour, it's a straightforward change to make ...

rosek86, 2016-02-02:
@sandeepmistry sorry for late reply with this, this is how this looks with different manufacturer identifiers compared to the same manufacturer identifiers: 

<5444010d ff62005c 03002100 00010000 00 554501 bf73fe1d cfcd> - 5444, then 5545
<54440100 ff62005c 03001f00 00010000 00          01 bf73 fe1dcfcd> - 5444, then 5444

iOS simply adds manufacturer data content if identifiers are the same, but when they're different it adds seconds manufacturer together with the scan response data.

BR, Krzysztof

sandeepmistry, 2016-02-21:
@rosek86 I've pushed https://github.com/sandeepmistry/noble/commit/7b633c6b5df80a81710c10abb6b877e89645a871 to the [combine-manufacturer-data branch](https://github.com/sandeepmistry/noble/tree/combine-manufacturer-data) for this behaviour on Linux.

Which version of OS X are you testing with? With noble on OS 10.11.3 I do not see OS X combining the manufacturer data ...

rosek86, 2016-02-22:
@sandeepmistry Did I mention that OS X combinating the manufacturer data? Please check my post from 24th Jan.

The OS X gives two separate buffers for advertisement and scan response, iOS merges these data into one buffer. I believe that the best way to handle manufacturer data is what iOS does,- combinate these buffers together. However two separate buffers would do too, as long as this is consistent for all platforms.

I think your commit looks good but you don't check manufacturer identifier before slicing it. I think this should look like this if we'd use iOS version:

    if (advertisement.manufacturerData) {
      var data = (bytes.readUInt16LE(0) === advertisement.manufacturerData.readUInt16LE(0) ? bytes.slice(2) : bytes);
      advertisement.manufacturerData = Buffer.concat([
        advertisement.manufacturerData,
        data
      ]);
    } else {
      advertisement.manufacturerData = bytes;
    }

And similar on OS X. I can have a look on OS X code and do a pull request for that if you like, but please confirm if you'd like to go with iOS version first.

sandeepmistry, 2016-02-27:
@rosek86 I had forgot to refer to the comments from earlier and was looking at the title. I will update the title accordingly to reflect this.

Your feedback to check for the company needs to be incorporated.

rzr, 2019-04-15:
Is there any traction for this change ?

Relate-to: https://github.com/abandonware/noble/issues/11
Relate-to: https://github.com/abandonware/noble/commit/7b633c6b5df80a81710c10abb6b877e89645a871