noble/noble #306: Unknown Peripheral with pimatic

tjandt, 2016-01-23:
Hi,

Im having a TI SensorTag CC2650 (http://www.ti.com/tool/cc2650stk)
and when trying to use it with pimatic and the pimatic-sensortag module (https://github.com/josecastroleon/pimatic-sensortag ... development branch) I'm getting the following error:

noble: unknown peripheral c4be8471c100 disconnected!
noble: unknown peripheral c4be8471c100 connected!

Linux  4.4.0-rc7+
Bluez version 5.36

Connecting via comandline works fine ... using the test scripts from your node_sensortag (test-cc2650-io.js) works as well. Any idea what to do?

sandeepmistry, 2016-01-23:
@tjandt I think there are multiple instances of noble running, one from ```node-sensortag``` and one from ```pimatic-ble```.

So when one connects, the other noble instance doesn't know about it and reports a warning. noble emits a ```warning``` event that can be used to suppress this output.

Can you please provide an HCI dump ```sudo hcidump -t -x```?

tjandt, 2016-01-26:
With pimatic running I get:

HCI sniffer - Bluetooth packet analyzer ver 5.36
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-01-26 19:46:55.250830 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)
2016-01-26 19:46:55.253071 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-01-26 19:46:55.268174 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x00 (disabled)
2016-01-26 19:46:55.270045 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-01-26 19:47:05.539513 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)
2016-01-26 19:47:05.543434 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-01-26 19:47:05.558050 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x00 (disabled)
2016-01-26 19:47:05.559448 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-01-26 19:47:15.828370 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)

sandeepmistry, 2016-01-30:
@tjandt I don't see any connection events in your capture. You'll need to run again, while the sensor tag is connected.

tjandt, 2016-01-30:
Okay sorry ... one more try with http://pastebin.com/9Kf2QSZ9

sandeepmistry, 2016-02-07:
@tjandt thanks for providing the capture, only one connection is made:

```
2016-01-30 22:29:29.017863 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role master
      bdaddr C4:BE:84:71:C1:00 (Public)
```

Here are a few options:
 * Make sure [pimatic-ble](https://github.com/josecastroleon/pimatic-ble/blob/master/package.json) is using the same version of noble as pimatic-sensortag
 * Add a ``noble.on('warning', function() { })`` listener in pimatic-ble
 * Only use pimatic-sensorbtag without pimatic-ble
 * Raise a new issue in noble, for removing the console.log for warning events

The warning has no impact on the system, it's just informing us that one noble instance (older version) does not know about a connection a newer noble instance is connecting to a peripheral.

closed 2016-02-07 by sandeepmistry