noble/noble #308: Make noble work with JXCore

marcnicholas, 2016-01-24:
I guess asking JXCore's "100% compatible with node" guarantee to extend to bindings is a bit much :)

Currently getting this:
```
dyld: lazy symbol binding failed: Symbol not found: __ZN4node6Buffer3NewEPKcm
  Referenced from: /Users/marc/Code/noble/node_modules/xpc-connection/build/Release/binding.node
  Expected in: dynamic lookup

dyld: Symbol not found: __ZN4node6Buffer3NewEPKcm
  Referenced from: /Users/marc/Code/noble/node_modules/xpc-connection/build/Release/binding.node
  Expected in: dynamic lookup

Trace/BPT trap: 5
```

I guess the version of node to bind to gets set in the gyp build process somewhere?

sandeepmistry, 2016-01-26:
> I guess the version of node to bind to gets set in the gyp build process somewhere?

Good question! For regular node you can pass a target parameter in: http://stackoverflow.com/a/15983893/2020087 (I have no idea what JXCore would need).

It feels like the build parameters are not right, for example it might be using Node.js headers but linking to JXCore? 

node-xpc-connection uses [```nan```](https://github.com/nodejs/nan) now, so doesn't use the the Node.js Buffer API's directly, but I imagine this line does something with Buffer new:

https://github.com/sandeepmistry/node-xpc-connection/blob/master/src/XpcConnection.cpp#L186

Maybe comment it out and rebuild to see if it's the root cause :) (as well as any other ```Nan::CopyBuffer``` lines you see ?

marcnicholas, 2016-01-26:
I edited the gyp config and got it to properly use jx (I think). But now seem to get some weird include errors...

```Marcs-MacBook-Pro:build marc$ make
  CXX(target) Release/obj.target/binding/src/XpcConnection.o
In file included from ../src/XpcConnection.cpp:5:
In file included from /Users/marc/.node-gyp/jxb310/src/node_buffer.h:8:
In file included from /Users/marc/.node-gyp/jxb310/src/jx/commons.h:34:
In file included from /Users/marc/.node-gyp/jxb310/src/jx/btree_map.h:31:
/Users/marc/.node-gyp/jxb310/src/jx/btree.h:919:10: warning: class 'btree_internal_locate_plain_compare' was previously declared as a struct
      [-Wmismatched-tags]
  friend class btree_internal_locate_plain_compare;
         ^
/Users/marc/.node-gyp/jxb310/src/jx/btree.h:893:8: note: previous use is here
struct btree_internal_locate_plain_compare {
       ^
/Users/marc/.node-gyp/jxb310/src/jx/btree.h:919:10: note: did you mean struct here?
  friend class btree_internal_locate_plain_compare;
         ^~~~~
         struct
/Users/marc/.node-gyp/jxb310/src/jx/btree.h:1188:8: error: expected member name or ';' after declaration specifiers
  void verify() const;
  ~~~~ ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/AssertMacros.h:1339:29: note: expanded
      from macro 'verify'
                #define verify(assertion) __Verify(assertion)
                                          ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/AssertMacros.h:420:5: note: expanded
      from macro '__Verify'
                  do                                                                      \
                  ^
In file included from ../src/XpcConnection.cpp:5:
In file included from /Users/marc/.node-gyp/jxb310/src/node_buffer.h:8:
In file included from /Users/marc/.node-gyp/jxb310/src/jx/commons.h:34:
In file included from /Users/marc/.node-gyp/jxb310/src/jx/btree_map.h:31:
/Users/marc/.node-gyp/jxb310/src/jx/btree.h:2049:16: error: expected unqualified-id
void btree<P>::verify() const {
               ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/AssertMacros.h:1339:29: note: expanded
      from macro 'verify'
                #define verify(assertion) __Verify(assertion)
                                          ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/AssertMacros.h:420:5: note: expanded
      from macro '__Verify'
                  do                                                                      \
                  ^
In file included from ../src/XpcConnection.cpp:5:
In file included from /Users/marc/.node-gyp/jxb310/src/node_buffer.h:8:
In file included from /Users/marc/.node-gyp/jxb310/src/jx/commons.h:34:
In file included from /Users/marc/.node-gyp/jxb310/src/jx/btree_map.h:32:

marcnicholas, 2016-01-27:
Couple of updates:

(1) Works fine on Linux with the HCI interface (good).
(2) The JX team figured out a [workaround](https://github.com/jxcore/jxcore/issues/802#issuecomment-175637609)

closed 2016-01-27 by marcnicholas

sandeepmistry, 2016-01-30:
@marcnicholas awesome news! Great to hear noble and it's dependencies work as is :)

You have it working on a MIPS board?

marcnicholas, 2016-01-30:
I have to setup a cross-compile gyp environment and possibly an OpenWRT package to get it to MIPS...but that's the plan.

sandeepmistry, 2016-01-30:
Cool, keep me posted.

Just a heads up there might be some issues with the endianness of the socket filter mask on MIPS, but I haven't confirmed it (https://github.com/sandeepmistry/noble/issues/294).

marcnicholas, 2016-01-30:
I've had it running on our MIPS board for a while under node.js. Broadcast works fine, but I've never been able to get a Peripheral mode connection established -- even with the native Bluez tools.

TheMaverickProgrammer, 2017-04-28:
where did you get the folder `~/.node-gyp/jxb310`? I've compiled jxcore from source. Have same issues as you did with Noble.