noble/noble #309: Question about the life cycle of the stateChange event

hgwood, 2016-01-24:
I'm writing new bindings for Noble so I can program my Sphero BB-8 using Windows 10's native bluetooth LE stack. I've got it working and I can control the robot. However, I have to trigger `noble.onDiscover` manually for the [Sphero SDK](https://github.com/orbotix/sphero.js) to pick up the peripheral, and I have trouble understanding how to implement scanning in order to get it to work properly. Here's a [code extract from the SDK](https://github.com/orbotix/sphero.js/blob/master/lib/adaptors/ble.js#L78) (`ble` is the noble instance):

```js
Adaptor.prototype.open = function open(callback) {
  var self = this;
  if (this.noblePeripheral != null) {
    // an existing noble peripheral was passed into adaptor
    self._connectPeripheral(this.noblePeripheral, callback);
  } else {
    // connect to peripheral using noble
    ble.on("discover", function(peripheral) {
      if (peripheral.id === self.uuid) {
        ble.stopScanning();
        self._connectPeripheral(peripheral, callback);
      }
    });

    ble.on("stateChange", function(state) {
      if (state === "poweredOn") {
        ble.startScanning();
      } else {
        ble.stopScanning();
      }
    });
  }
};
```

In the "normal" case, `this.noblePeripheral` is `null` (i.e. not set manually, we want to discover it) so the `else` part is run. I understand the `ble.on("discover"` part, but the rest puzzles me. There is nothing in the `else` that triggers anything in `ble`. It only reacts. So how can this code be sure that a `stateChange` event will ever be triggered? What if `ble` is already `poweredOn` when `open` is called? Or is it that `stateChange` is something that happens regularly? How can my bindings know that they are expected to "power on" at some point? Is this code even using Noble correctly?

Thanks in advance for any help you could provide.

hgwood, 2016-01-24:
This piece of code + an idempotent `startScanning` implementation does the trick but I'd like to understand more.

```js
NobleBindings.prototype.init = function() {
  var self = this;
  setInterval(function () {
    self.emit('stateChange', 'poweredOn');
  }, 1000);
};
```

sandeepmistry, 2016-01-26:
@hgwood the state change event just needs to be triggered once, every app or lib that uses noble needs to wait for a powered on event before using any of the noble API's.

If you're sure the adapter will always be powered on in your situation, triggering it once on the next tick is fine:

```javascript
NobleBindings.prototype.init = function() {
  process.nextTick(function() {
     this.emit('stateChange', 'poweredOn');
  }.bind(this));
};
```

Side note, please let me know if you have your code online in the future.

I'm closing this for now, feel free to re-open if I haven't answered your question or if something is unclear.

closed 2016-01-26 by sandeepmistry

hgwood, 2016-01-26:
Thanks, it makes sense. Would I be correct then, if I said that the code extract I gave is not 100% correct because the `poweredOn` event could have already been fired? Instead the code should read:

```js
  if (ble.state === "poweredOn") {
    ble.startScanning();
  } else {
    ble.on("stateChange", function(state) {
      if (state === "poweredOn") {
        ble.startScanning();
      } else {
        ble.stopScanning();
      }
    });
  }

```

hgwood, 2016-01-26:
Oh and I'll definitely push those bindings to GitHub. :)

sandeepmistry, 2016-01-30:
@hgwood great!

Yes, I agree with your statement in https://github.com/sandeepmistry/noble/issues/309#issuecomment-175210464.

hgwood, 2016-02-19:
Here are my bindings: https://github.com/hgwood/winble. I'm planning on publishing them to npm and then come back here and open an issue to see if we could work together to make noble work on Windows 10 out-of-the-box.

sandeepmistry, 2016-02-21:
@hgwood this is awesome, thanks for sharing. I'll repost in another Windows related thread.