noble/noble #310: add support for the bluez 5.x dbus api

jrobeson, 2016-01-24:
I see that you plan on using the dbus API in the future based on this comment https://github.com/sandeepmistry/noble/issues/115#issuecomment-67829758 , but are holding off because of bluez 4.x still being supported. Is there some way in which it is reasonably possible to support both? A runtime switch (while still building bluetooth-hci-socket) would be sub-optimal, but at least it would give us the ability to avoid  sudo or CAP_NET_ADMIN requirements.

jrobeson, 2016-01-24:
I also realized that a lot of folks tend to avoid dbus in distros like Gentoo or on server boxes, so one would still want the bluetooth-hci-socket module to work as a fallback anyways.

jrobeson, 2016-01-24:
I didn't realize you'd already started something for this effort (as seen here): https://github.com/sidorares/node-dbus/issues/13

sandeepmistry, 2016-01-26:
@jrobeson do you have time to look at this?

One note the some of the D-Bus API's are still marked as "experimental", I'm not sure if the Debian Jessie release for example, builds with them enabled and starts bluetoothd with -E argument

 * http://git.kernel.org/cgit/bluetooth/bluez.git/tree/doc/adapter-api.txt
 * http://git.kernel.org/cgit/bluetooth/bluez.git/tree/doc/gatt-api.txt
 * http://git.kernel.org/cgit/bluetooth/bluez.git/tree/doc/device-api.txt
 * http://git.kernel.org/cgit/bluetooth/bluez.git/tree/doc/advertising-api.txt

There's also a really old branch: https://github.com/sandeepmistry/noble/tree/linux-bluez5

FWIW, the Node.js D-Bus libs seemed a little rough at the time ...

jrobeson, 2016-01-26:
yeah, I can't imagine it being the default implementation for some time to come.

I will be diving into your code at least a little bit tonight, so thanks!

jrobeson, 2016-01-26:
so why did you switch from dbus-native exactly?

jrobeson, 2016-01-26:
with a few minor changes (mostly to get semi up to speed with recent changes in the codebase), I was able to run  `examples/advertisement-discovery.js`.

```sh
$ node advertisement-discovery.js
peripheral discovered (c406e7c75d5d with address <C4:06:E7:C7:5D:5D, null>, connectable null, RSSI -42:
	hello my local name is:
		HRM1
	can I interest you in any of the following advertised services:
		["180d","180a"]
```

I hven't yet figured out how (or if possible) to get connectable and address type working

jrobeson, 2016-01-26:
Oh, I should metntion that I removed the commit that switched from dbus-native, mostly because it has > 0 tests, and has seemingly responsive maintainer.  I can switch back to the other module if you really prefer it though.

sandeepmistry, 2016-01-30:
@jrobeson is your work up on Github?

> I hven't yet figured out how (or if possible) to get connectable and address type working

OS X currently returns "unknown" for the address type, a similar approach can be taken.

> Oh, I should mention that I removed the commit that switched from dbus-native. dbus-native has > 0 tests, and a seemingly responsive maintainer. We can switch back to the other one if you really prefer it though.

No preference on this, what ever works and is stable.

jrobeson, 2016-01-30:
my working isn't public yet, because it's not even close to correct. I'm also hardly familiar with node, and only really understand dbus on a conceptual level.

I made it work in the code you had in InterfacesAdded, but that only works in cases where we don't know about any devices yet. As soon as we find them the first time, this event will no longer fire.

As soon as i rework it around PropertiesChanged, then I will will publlish it.

sandeepmistry, 2016-06-06:
@jrobeson any progress on this?

jrobeson, 2016-06-06:
I made some, and then i decided to try out without so many nested callbacks, sorta like this https://github.com/sidorares/node-gday/blob/master/lib/index.js 

After that I got moved to another project, but am about to have time open up for this project again soon.   If you wanna give it a try instead of waiting, then you're certainly welcome to it.

sandeepmistry, 2016-06-19:
Thanks for the update. I'm going to hold off working on this myself, so it's still up for grab if someone else wants to work on it.