noble/noble #313: Double ServiceUUID bug

paulgavrikov, 2016-01-26:
Hi Sandeep,

I have been using you lib for 2 months now and added bunch of functions (hci / linux only). I also found a bug: When using OS X devices, some apps push services like '180a' which are already provided by OS X or iOS Gatt. Which means that if one discovers services he will get that service twice. Note that those services might have totally different characteristics. Since noble is not using handles, discovering Characteristics of a Services that is non-unique will only discover characteristics of one of those non-unique services.

I totally agree that this is against the BLE Spec, but it happens occasionally. E.g. when launching virtual Peripherals from the LightBlue iOS app.

This might be a minor problem but also causes a bug where ```discoverAllServicesAndCharacteristics ``` or ``` disocverSomeServicesAndCharacteristics``` will get stuck. 

I patched this in ```noble/lib/hci/peripheral.js``` but I don't know how to create a pull request so I post it here. Note: this only helps the function to terminate, so it's a workaround for off-spec BLE Devices.

```
Peripheral.prototype.discoverSomeServicesAndCharacteristics = function(serviceUuids, characteristicsUuids, callback) {
  this.discoverServices(serviceUuids, function(err, services) {

    var servicesLength = services.length;
    var knownUUIDs = [];

    for (var s in services) { 
      var service = services[s];
      if(knownUUIDs.indexOf(service.uuid) != -1)
        servicesLength--;
      knownUUIDs = knownUUIDs.concat(service.uuid);
    }

    var numDiscovered = 0;
    var allCharacteristics = [];

    for (var i in services) {
      var service = services[i];

      service.discoverCharacteristics(characteristicsUuids, function(error, characteristics) {
        

        numDiscovered++;

        if (error === null) {
          for (var j in characteristics) {
            var characteristic = characteristics[j];

            allCharacteristics.push(characteristic);
          }
        } 

        if (numDiscovered === servicesLength) {
          if (callback) {
            callback(null, services, allCharacteristics);
          }
        }
      }.bind(this));
    }
  }.bind(this));
};
```

jrobeson, 2016-01-26:
creating a pull request is easy in the web editor. Just navigate to the file and click the edit button.

sandeepmistry, 2016-01-30:
@paulgavrikov thanks for reporting.

> I totally agree that this is against the BLE Spec, but it happens occasionally. E.g. when launching virtual Peripherals from the LightBlue iOS app.

Actually it's allowed, I think it would be better to fix #175 than working around it.

I've done some work to add id's to attributes: https://github.com/sandeepmistry/noble/tree/attribute-ids

However, the distributed and web socket bindings still need changes. On a side note, I've been thinking about removing them, and releasing a 2.0 breaking change release.

paulgavrikov, 2016-01-30:
Let me quote Bluetooth Core Specification 4.0  --  3.1 Service Definition: 

> When multiple services exist, services definitions with service declarations using 16-bit Bluetooth UUID should be grouped together (i.e. listed sequentially) and services definitions with service declarations using 128-bit UUID should be grouped together.

So two similar UUIDs should be grouped, anyway #318 is a hotfix one could consider until IDs are used. 

Thanks for your work!

paulgavrikov, 2016-01-30:
You are right, the BLE Spec allows that (for what reason ever). If someone wants to use a workaround until noble switches to IDs: #318

closed 2016-01-30 by paulgavrikov