noble/noble #314: System Stalls After X amount of time

mahdiinaya, 2016-01-26:
I'm using NOBLE source code to scan for multiple beacons over long time periods. My code is very similar to the "enter-exit.js" example. Though, I continue printing out to console all iBeacons that I see regardless if I have seen them before or not.

At first, the source code shows to work correctly in that I am able to see correct data. After some given time (anywhere from 20min~3hours), the code stalls out for some reason and we stop seeing any inputs.

No errors are thrown. The system just stalls. After many hours of trying to find out the issue, I am still at a loss.

What can be done to fix this situation?

jrobeson, 2016-01-26:
you'd need to mention your OS and your bluez version too.

mahdiinaya, 2016-01-27:
Oh very sorry.

I'm running on a Raspberry Pi Model B+. Wheezy 8.0. Bluez version: 5.23-2+b1

I'm not sure what to trace or even where to look. My only solution so far is a reset every so often 

```javascript
setInterval(function resetSystem() {
//debugger;
noble.stopScanning();
console.log('SYSTEM RESET BEGIN');
noble.startScanning([], true);
console.log('SYSTEM BACK UP');
}, resetSystemInterval);```

sandeepmistry, 2016-02-07:
@mahdiinaya please provide an HCI dump capture (via Github Gist, or pastebin).

```sh
sudo hcidump -t -x
```

mahdiinaya, 2016-02-08:
@jrobeson 

**Here is a small snippet of the strace while everything is working fine:**

stat64("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3519, ...}) = 0
write(12, "IBEACON FOUND:Tue Jan 26 2016 21"..., 97) = 97
clock_gettime(CLOCK_MONOTONIC, {3576, 803711774}) = 0
clock_gettime(CLOCK_MONOTONIC, {3576, 804313767}) = 0
epoll_wait(5, {{EPOLLIN, {u32=11, u64=11}}}, 1024, 0) = 1
clock_gettime(CLOCK_MONOTONIC, {3576, 805585752}) = 0
read(11, "\4>#\2\1\0\1\346\370\336\254pE\27\2\1\6\23\377L\0\f\16\0\233\257[\234\312\313\370\34"...,
1024) = 38
clock_gettime(CLOCK_MONOTONIC, {3576, 812316671}) = 0
clock_gettime(CLOCK_MONOTONIC, {3576, 812851665}) = 0
epoll_wait(5, {{EPOLLIN, {u32=11, u64=11}}}, 1024, 488) = 1
clock_gettime(CLOCK_MONOTONIC, {3576, 814088650}) = 0
read(11, "\4>\f\2\1\4\1\346\370\336\254pE\0\266", 1024) = 15
clock_gettime(CLOCK_MONOTONIC, {3576, 829616465}) = 0
clock_gettime(CLOCK_MONOTONIC, {3576, 830696452}) = 0
epoll_wait(5, {{EPOLLIN, {u32=11, u64=11}}}, 1024, 470) = 1
clock_gettime(CLOCK_MONOTONIC, {3576, 832128435}) = 0
read(11, "\4>*\2\1\0\1\10\315=\3670\357\36\2\1\6\32\377L\0\2\25\271@\1770\365\370Fn\257"...,
1024) = 45

**Here it is when it stops working:**
ioctl(11, 0x800448d3, 0xbeb56fbc)       = 0
clock_gettime(CLOCK_MONOTONIC, {5800, 79300606}) = 0
clock_gettime(CLOCK_MONOTONIC, {5800, 80306595}) = 0
epoll_wait(5, {}, 1024, 1001)           = 0
clock_gettime(CLOCK_MONOTONIC, {5801, 83551711}) = 0
clock_gettime(CLOCK_MONOTONIC, {5801, 84073706}) = 0
clock_gettime(CLOCK_MONOTONIC, {5801, 84721699}) = 0
ioctl(11, 0x800448d3, 0xbeb56fbc)       = 0
clock_gettime(CLOCK_MONOTONIC, {5801, 86729677}) = 0
clock_gettime(CLOCK_MONOTONIC, {5801, 87474669}) = 0
epoll_wait(5, {}, 1024, 1001)           = 0

mahdiinaya, 2016-02-08:
@sandeepmistry below is the hcidump. Please note that while it was working, everything shows to be working fine. The bluetooth dongle seems to pull in data nicely. The time is now 2016-02-08 18:30:00, the last input to the bluetooth dongle was as shown : 2016-02-08 12:59:55 The bluetooth dongle just stops receiving data. It is as if it goes to sleep. 

2016-02-08 12:59:55.180591 > HCI Event: LE Meta Event (0x3e) plen 37
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr CB:D4:67:31:25:91 (Random)
      Complete local name: 'estimote'
      Unknown type 0x16 with 13 bytes data
      RSSI: -91
2016-02-08 12:59:55.234600 > HCI Event: LE Meta Event (0x3e) plen 42
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr C9:1A:8B:A0:CB:CA (Random)
      Flags: 0x06
      Unknown type 0xff with 25 bytes data
      RSSI: -65
2016-02-08 12:59:55.237614 > HCI Event: LE Meta Event (0x3e) plen 37
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr C9:1A:8B:A0:CB:CA (Random)
      Complete local name: 'estimote'
      Unknown type 0x16 with 13 bytes data
      RSSI: -59
2016-02-08 12:59:55.240598 > HCI Event: LE Meta Event (0x3e) plen 42
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr F6:73:FC:C6:79:66 (Random)
      Flags: 0x06
      Unknown type 0xff with 25 bytes data
      RSSI: -78
2016-02-08 12:59:55.243611 > HCI Event: LE Meta Event (0x3e) plen 37
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr F6:73:FC:C6:79:66 (Random)
      Complete local name: 'estimote'
      Unknown type 0x16 with 13 bytes data
      RSSI: -78

sandeepmistry, 2016-02-13:
@mahdiinaya do you have a HCI dump from start to end instead of just a snippet?

I don't see anything in this snippet to indicate a stop scanning command was sent, so it could very well be a firmware issue with the adapter it self. When did you start scanning, when do things stop working, are the results consistent?

sandeepmistry, 2016-02-28:
Closing due to lack of activity, please re-open with requested info if you are still interested.

closed 2016-02-28 by sandeepmistry