noble/noble #317: add a .travis.yml for travis-ci

jrobeson, 2016-01-27:
It'd be nice to have travis run all tests for PRs/pushes.

jrobeson, 2016-01-27:
I already gave this a try on my github fork, and we already have a build failure on nodejs 0.8.x If this failure isn't transient, then it seems like we should update the minimum node engine version in package.json.

You can see the the build failures here: https://travis-ci.org/jrobeson/noble/builds/105102613

Here's the .travis.yml I'm using (to save a clickover to the repo)

```yml
language: node_js
node_js:
  - "5"
  - "4.1"
  - "4.0"
  - "0.12"
  - "0.11"
  - "0.10"
  - "0.8"
env:
  - CXX=g++-4.8
addons:
  apt:
    sources:
      - ubuntu-toolchain-r-test
    packages:
      - g++-4.8
```

jrobeson, 2016-01-27:
I see i missed 4.2, so you might wanna add that. Alternatively,  if node has been good at keeping semver for 4.x, then we can just use "4". I'm not experienced enough with node stuff to be able to tell.

sandeepmistry, 2016-01-30:
For node 0.8 looks like ```date-now``` is not available, it's a sub-dependency of the jshint dev dep.

I'm not opposed to removing support for 0.8, one thing to try first is upgrading the jshint dev dep version to the latest release, or making sure a later npm version is used.

Does travis support OS X as well?

jrobeson, 2016-01-30:
travis supports osx, but only 10.9.5.  I don't think it matters too much though, unless you know how to setup virtual BLE  devices on there that we could run functional tests against.

On another note, Appveyor exists to run CI tests on windows, I"m just not familiar with it at all.

jrobeson, 2016-01-31:
Do you find any value in testing on OSX without functional tests? I don't wanna load travis unnecessarily nor cause test times to double, if we don't really get any benefit from it.

jrobeson, 2016-01-31:
Oh, in regards to jshint. They don't even test against 0.8.x at all. You can see that here: https://github.com/jshint/jshint/blob/master/.travis.yml

sandeepmistry, 2016-02-07:
> Do you find any value in testing on OSX without functional tests? 

Yes, OS X has different dependencies than Linux and Windows (xpc-connection) it would be valuable to see if they can be installed and built.

Similar to how you discovered a devDep issue with Node.js 0.8.x.

jrobeson, 2016-02-07:
I'll give it a go.

jrobeson, 2016-02-20:
I found out that nodejs builds are not available for mac osx. Says so here: https://docs.travis-ci.com/user/languages/javascript-with-nodejs/

jrobeson, 2016-02-20:
actually, you can support OSX, but it does seem like a lot of work. Here's one example: https://github.com/EvolveLabs/electron-celt/blob/master/.travis.yml

jrobeson, 2016-02-20:
I actually got something much simpler to work now! https://travis-ci.org/jrobeson/noble . Everything builds for both mac and linux (except 0.8).   If you want to get 0.8 working, then please tell give me a specific action to try, because I'm not really interested in spending time on anything that old.

jrobeson, 2016-02-20:
A link to the new config file: https://github.com/jrobeson/noble/blob/add-travis-ci/.travis.yml

sandeepmistry, 2016-02-21:
@jrobeson nice work!

So where do we go from here to get this in noble master?

jrobeson, 2016-02-21:
need to decide on .8 first,  having a build that will be marked as failed and letting it continue to fail, mostly makes it moot. it will show all PRs as failed atm until that is solved

jrobeson, 2016-02-21:
once that is done, then you'll need to follow steps 1 and 2 here: https://docs.travis-ci.com/user/getting-started/  . I will submit the PR then.

sandeepmistry, 2016-02-27:
@jrobeson my preference for this is to leave package.json alone for now (with 0.8 support) and only have travis ci run for Node.js version 0.10 and above.

At the next major release (v2.0) we can remove 0.8 support from noble.

jrobeson, 2016-02-27:
but are you sure that 0.8 even works for anybody? if it isn't even installable, then why bother?

jrobeson, 2016-02-27:
let me know when you've followed the instructions i mentioned in https://github.com/sandeepmistry/noble/issues/317#issuecomment-186921406 and then I will submit the PR

sandeepmistry, 2016-02-27:
v0.8.28 with an upgraded npm works on my Mac. The bundled version of npm (1.2.30) with 0.8.28, doesn't seem to support ```^x.y.z``` version syntax. This is why the Travis Ci build was failing.

```
Sandeep-Mistrys-MacBook-Air:noble-node-0.8 smistry$ node -v
v0.8.28
Sandeep-Mistrys-MacBook-Air:noble-node-0.8 smistry$ npm install noble
npm WARN optional dep failed, continuing bluetooth-hci-socket@0.4.2
-
> xpc-connection@0.1.4 install /Users/smistry/Developer/noble-node-0.8/node_modules/noble/node_modules/xpc-connection
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
  SOLINK_MODULE(target) Release/binding.node
noble@1.3.0 node_modules/noble
├── bplist-parser@0.0.6
├── debug@2.2.0 (ms@0.7.1)
└── xpc-connection@0.1.4 (nan@2.2.0)
Sandeep-Mistrys-MacBook-Air:noble-node-0.8 smistry$ node node_modules/noble/examples/advertisement-discovery.js 
peripheral discovered (a0b051b9bed14d0bbc5f982c6784f5c0 with address <unknown, unknown>, connectable true, RSSI -59:
	hello my local name is:
		Apple TV
	can I interest you in any of the following advertised services:
		[]
	here is my manufacturer data:
		"4c000906030a0a00012b"
```

sandeepmistry, 2016-02-27:
Steps 1 and 2 completed on Travis ....

jrobeson, 2016-02-28:
fixed in #342

closed 2016-02-28 by jrobeson