noble/noble #323: My Programm works with an old noble version but not with the newest one.

davidmann4, 2016-01-30:
This uses the version "noble": "1.0.x"
[image]

lets switch to "noble": "1.0.x"
[image]

```javascript
var noble = require('noble');

var _characteristic = null;

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    noble.startScanning(['ffe0'], false);
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', function(peripheral) {
  console.log('peripheral discovered (' + peripheral.id +
              ' with address <' + peripheral.address +  ', ' + peripheral.addressType + '>,' +
              ' connectable ' + peripheral.connectable + ',' +
              ' RSSI ' + peripheral.rssi + ':');
  console.log('\thello my local name is:');
  console.log('\t\t' + peripheral.advertisement.localName);
  console.log('\tcan I interest you in any of the following advertised services:');
  console.log('\t\t' + JSON.stringify(peripheral.advertisement.serviceUuids));

  peripheral.connect(
    function (error) {
      console.log(error);
    }
  );

  peripheral.discoverSomeServicesAndCharacteristics(['ffe0'], [],
                function (err, services, characteristics) {

      _characteristic = characteristics[0];
      characteristics[0].on('read', function (data, isNotification) {
        // Push the data into the stream buffer
        console.log(data);
      });
  });

  var serviceData = peripheral.advertisement.serviceData;
  if (serviceData && serviceData.length) {
    console.log('\there is my service data:');
    for (var i in serviceData) {
      console.log('\t\t' + JSON.stringify(serviceData[i].uuid) + ': ' + JSON.stringify(serviceData[i].data.toString('hex')));
    }
  }
 
  var buf = "REMOVED CODE HERE TO KEEP IT SIMPLE - GENERATED A BYTE ARRAY";
  setTimeout(function(){ writeBLEdata(buf); }, 3000);
});

function writeBLEdata(chunk) {
  // Do we have a characteristic?
  if (_characteristic) {
    // Then write our data to it, in 20 byte chunks
    for (var i = 0; i < chunk.length; i += 20) {
      _characteristic.write(chunk.slice(i, i + 20), true);
    }
    // Everything is fine
    console.log("wrote some data");
  } else {
    // There's a problem
    console.log("No connection");
  }
}
```

davidmann4, 2016-01-30:
based on https://github.com/xorbit/node-hm-ble/issues/1 - would like to see something like this in the example folder. BLE UART is kinda popular.

sandeepmistry, 2016-02-07:
@davidmann4 what is printing out the "Error: command disallowed" message?

Please also provide an HCI dump capture: ```sudo hcidump -t -x```

davidmann4, 2016-02-07:
peripheral.connect(
    function (error) {
      console.log(error);
    }
  );

can provide  "sudo hcidump -t -x" output later used a ubuntu VM and a 10$ usb ble stick which worked with the older version

sandeepmistry, 2016-02-21:
Closing for now due to inactivity, please re-open with requested info. is you are still interested.

closed 2016-02-21 by sandeepmistry