noble/noble #324: Add a way to set the adapter without an ENV var.

jrobeson, 2016-01-30:
Since you mentioned 2.0 being on the horizon in https://github.com/sandeepmistry/noble/issues/313#issuecomment-177288884, then it might be a good idea to change the public API of noble so that adapters could be set in the Noble constructor or a settter.

It might also be a good time to add a `listAdapters` method.

sandeepmistry, 2016-02-07:
@jrobeson I'm not too keen about this, OS X would not support this API (see [CoreBluetooth API](https://developer.apple.com/library/mac/documentation/CoreBluetooth/Reference/CoreBluetooth_Framework/)).

I think most users systems only have a single Bluetooth adapter as well.

sandeepmistry, 2016-02-07:
Closing this, as there is a similar discussion in #325.

closed 2016-02-07 by sandeepmistry