noble/noble #325: Potential API Change for 2.x

jrobeson, 2016-01-30:
If 2.x is coming, then it might be time to think about  making devices  tied to the connected adapter.

So noble would work like this instead.

```js
var noble = require('noble');

var adapterPathOrObject= 'actualDeviceNameOrObject'; // not sure which one you'd prefer (or both)
var adapter = noble.getAdapter('AdapterPathOr)
adapter.startScan(); // 

var devices = adapter.getDevices();
var device = devices[perpipheralObjectOrId];
device.getServices();  // or characteristics, attributes like RSSI, or anything else
```

This is just just playing around though, I don't actually know what would be the best.

sandeepmistry, 2016-02-07:
See comment in https://github.com/sandeepmistry/noble/issues/324#issuecomment-181120873.

rzr, 2021-03-02:
Please can you update status of this issue on this latest release: 

https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-12