noble/noble #326: perhaps add clearCachedData method

jrobeson, 2016-01-30:
In my dbus experiments, I've found https://github.com/adafruit/Adafruit_Python_BluefruitLE . They have a method to clear the cached bluetooth data, and I wonder if folks would find that method useful in noble.

Here's a link to the CoreBluetooth provider (aka binding) https://github.com/adafruit/Adafruit_Python_BluefruitLE/blob/master/Adafruit_BluefruitLE/corebluetooth/provider.py#L300

sandeepmistry, 2016-02-07:
That functionality it rarely needed and could potential mess up someone's Mac.

closed 2016-02-07 by sandeepmistry

ppannuto, 2016-02-10:
I think there may be some need for this, I just got bit by this today and spent a while trying to figure out what's going on.

This is the current behavior of noble 1.3.0 on my mac:

1. Program a BLE node to advertise "Name1"
2. Run noble script to list names (below), reports "Name1"
3. Kill noble script
4. Re-program device to advertise "Name2"
5. Run noble script, reports "Name1"
6. Kill noble script
7. Run noble script, reports "Name2"

This is consistent, reproducible, and very confusing.

```javascript
var noble = require('noble');

noble.on('stateChange', function (state) {
	if (state === 'poweredOn') {  noble.startScanning();  }
});

noble.on('discover', function (peripheral) {
	console.log('Found: ' + peripheral.advertisement.localName);
});
```

benjaminhon, 2017-06-19:
when a peripheral state on connected has state == 'error' , i do need to clear the cache else it never ever detects again, can i just do
```
delete noble._peripherals[uuid] ?
```