noble/noble #329: Troubleshooting "noble warning: unknown peripheral xxxxxxxxxxxx" on characteristic.read

jpodonn, 2016-02-09:
I have a custom BLE peripheral that I am successfully reading data from with an Intel Edison. From the given examples and my pretty limited knowledge of javascript I was able to hack together a program that connects, discovers my UUIDs, reads 7 characteristics at predetermined intervals, and is notified and reads another. 

My problem is that after a considerable length of time, I start getting an unknown peripheral warning instead of reading data. The unknown peripheral warning continues until I stop the progream. If I stop the program and restart it, it will start reading data again, but I need this to happen automatically because I need to be collecting data 24/7. 

So I need to know either how to troubleshoot this issue or how to disconnect, reconnect, and resume reading data if this warning occurs. 

Any help in this would be greatly appreciated. Noble has saved me a huge amount of time already and I hope to be able to use it throughout my project.

sandeepmistry, 2016-02-13:
@jpodonn please provide and HCI dump capture: ```sudo hcidump -t -x```

jpodonn, 2016-02-19:
Thanks for your response, I will get the HCI dump when I am back at the office Monday.

jpodonn, 2016-02-23:
I have attached the full HCI dump. 
This is from an example that worked for just a couple of minutes. It appears that this is where the problem occurs:

2016-02-23 17:36:48.571911 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x08
    Reason: Connection Timeout

I'm not sure how to handle this. I need to restart the connection after a timeout.

[hcidump.txt](https://github.com/sandeepmistry/noble/files/143072/hcidump.txt)

sandeepmistry, 2016-02-27:
@jpodonn thanks for the capture, unfortunately I only see a single connection event and disconnect event, nothing that indicates an unknown peripheral has been connected:

```
...

2016-02-23 17:32:03.003810 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role master
      bdaddr C4:BE:84:E9:57:09 (Public)
...

2016-02-23 17:36:48.571911 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x08
    Reason: Connection Timeout
```

As for next steps, please provide the following:

1) Minimal amount to sample code to reproduce the issue
2) Debug output from running 1): DEBUG=* node <file>.js
3) Another HCI dump capture while running 2)

sandeepmistry, 2016-03-13:
Closing due to lack of feedback, please re-open with requested information if you are still interested.

closed 2016-03-13 by sandeepmistry

vinu123456789, 2017-10-30:
I have the same issue of noble warning.even i want to make my application run automatically for 24/7 but the issue is if my device is disconnecting suddenly during the "read" then iam not able to come out of this noble warning even though i had setup noble warning callback to close the application so i can restart my application  by putting in while loop .

so please help me in this issue.

Continuous warning msg on console : 
[image]

 HCI dump Output : sudo hcidump -t -x 

[image]

jurstyx, 2020-01-07:
Hello, I have the same issue as @vinu123456789. Does somebody have a solution to prevent the application stopping after the noble exceptions?