noble/noble #331: Same code that runs on OSX wont run on ubuntu

dorshay6, 2016-02-11:
Hi,

I have wrote a simple code that reads and write from beacons.
The code dveloped on my MBP and runs perfect in also on MBA both with different  OSX versions.
When i try to run the code on ubuntu pc's the code don't return the same service that my MBP finds.

`noble.on('discover', function(peripheral)
{
    var peripheralID = uuid.v4()
    worker.found.push({id:peripheralID,status:-2});
    activeBeacons.push({id:peripheralID,obj:peripheral});
    wss.boradcastWorker();

    //console.log('Found device with local name: ',peripheral);

    if(worker.status != 1)
    {
        return;
    }

    peripheral.connect(function(error)
    {
        setBeaconsArrAttr(peripheralID,'isConnect',1)
        setBeaconsArrAttr(peripheralID,'status',-1)

        peripheral.discoverServices(['FFF0'], function(error, services)
        {
            //Fail point on ubuntu <--
            if(!services[0])
                return;

            var deviceInformationService = services[0];

            deviceInformationService.discoverCharacteristics(['FFF7','FFF5','FFF1','FFF2','FFF3'], function(error, characteristics)
            {
                if(!characteristics[4])
                {
                    return;
                }

                var pass = characteristics[4]
                pass.write(new Buffer('AA140611','hex'), false, function(x,y)
                {
                    console.log(new Buffer('AA140611','hex'))
                });

                pass.on('write',function(x,y)
                {
                    getBeaconData (peripheralID,characteristics)
                });
            });
        });
    });
});`

jrobeson, 2016-02-13:
you need to mention your OS version. that could pin it down to a bug in the bluez version. I'm not saying that it definitely is, but it is helpful information

sandeepmistry, 2016-02-13:
@dorshay6 can you try with lower case UUID's? For example: ```fff0``` instead of ```FFF0```.

dorshay6, 2016-02-14:
@sandeepmistry That was the problem! Tanks!

sandeepmistry, 2016-02-21:
@dorshay6 ok good, to know. Let's keep this issue open, I'll renamed accordingly.

jacobrosenthal, 2016-09-21:
Id like to see us standardize uuids by testing and format inputs in the 'abstract' were working on in https://github.com/sandeepmistry/noble/pull/431

sandeepmistry, 2016-09-21:
@jacobrosenthal agreed.

I've been thinking we could create a custom UUID type that extends `String`.