noble/noble #332: kCBAdvDataIsConnectable and Manufacturer Data are cached on MacOS X Yosemite

mdlampi, 2016-02-12:
It appears that kCBAdvDataIsConnectable and Manufacturer Data are cached on MacOS X Yosemite in Noble. The Mac Bluetooth Explorer shows that these values are 0 and missing when my target device is paired and actively talking to a separate process on the Mac (via Noble), yet the values are present and available to a second process (also using Noble).

No matter what I try, e.g., telling Noble to stop scanning and start scanning, seems to get Noble to flush these values.

The reported services and RSSI values are updated, however. The services count goes to zero (0) when the first process is talking to the device, and returns to one (1) after the first process disconnects from the device.

Here is the code in coffeescript that I use to check the values of mfgdata and connectable:

bunyan = require('bunyan')
logger = require('../src/logger')

logger.level('debug')
noble = require('noble')
###
  Scans for peripherals and prints out message when we find a particular device
    In range criteria:      RSSI < threshold
###

RSSI_THRESHOLD = -90
localName = ''
serviceUuid = '87654321890123456789012345678900'
streamChar = readChar = writeChar = null
circulatorAddress = process.env.CIRCULATOR_ADDRESS
if not circulatorAddress
  throw new Error("You must set the CIRCULATOR_ADDRESS environment variable before running tests")
circulator = '5901' + circulatorAddress

noble.on 'discover', (peripheral) ->

  return unless peripheral.rssi > RSSI_THRESHOLD

  # return unless peripheral.advertisement.serviceUuids.length isnt 0

  if localName is ''
    return unless peripheral.advertisement.manufacturerData?

    mfgAddress = peripheral.advertisement.manufacturerData.toString('hex')
    return unless mfgAddress is circulator
    logger.debug 'peripheral discovered (' + peripheral.id + ')' + \
                  ' with address <' + peripheral.address +  ', ' + peripheral.addressType + '>,' + \
                  ' connectable ' + peripheral.connectable + ',' + \
                  ' RSSI ' + peripheral.rssi + ':'
    logger.debug '\tMy local name is:\t' + peripheral.advertisement.localName
    localName = peripheral.advertisement.localName

    return unless peripheral.advertisement.serviceUuids?
    return unless peripheral.advertisement.serviceUuids.length isnt 0
    logger.debug '\tMy advertised services:\t' + JSON.stringify(peripheral.advertisement.serviceUuids)
    return unless peripheral.advertisement.serviceUuids[0] is serviceUuid

  else
    #logger.debug "Not my device:", peripheral.advertisement.localName
    return if peripheral.advertisement.localName isnt localName

  # we found a peripheral, stop scanning
  noble.stopScanning()

  if peripheral.advertisement.manufacturerData?
    logger.debug 'My manufacturer data:\t' + JSON.stringify(peripheral.advertisement.manufacturerData.toString('hex'))

  if peripheral.connectable
    logger.debug "#{localName}/#{peripheral.rssi} is connectable: #{peripheral.connectable}"
    logger.debug "Services:  #{peripheral.advertisement.serviceUuids.length}"
  else
    logger.debug "***#{localName}/#{peripheral.rssi} is not connectable: #{peripheral.connectable}***"
    logger.debug "Services:  #{peripheral.advertisement.serviceUuids.length}"

  # we want to start scanning again
  noble.startScanning [], true

noble.on 'stateChange', (state) ->
  if state == 'poweredOn'
    logger.debug "Starting to scan"
    noble.startScanning [], true
  else
    noble.stopScanning()

sandeepmistry, 2016-02-21:
@mdlampi what output do you get from your app and what are you expecting?

noble currently "ors" advertisement data values, so removed values will not be cleared (https://github.com/sandeepmistry/noble/blob/master/lib/noble.js#L136-L140).

Also, debug output would be useful: ```DEBUG=* node <file>.js```

mdlampi, 2016-02-22:
Hello Sandeep,

I am expecting that the library should return that the device is not
connectable when that device is connected to a different host.
Unfortunately, it returns that the device is connectable even when other
tools report the opposite to be true.

Attached is a log of a test run on Mac OS X Yosemite on an iMac.

Note that at the beginning of the test run the library is correctly showing
that the device is connectable, yet later on when the device is paired and
actively being used by a different host the library still shows the device
as being connectable.

Best regards,

Michael Lampi

sandeepmistry, 2016-02-27:
@mdlampi I am closing this, as the connectable flag is based on GAP data, it is not suppose to update when the peripheral is connected to another host.

closed 2016-02-27 by sandeepmistry