noble/noble #333: Can I skip encription procedure for TK and STK?

BYAKheee, 2016-02-15:
bleno/noble are awesome library. I’ｍ using it on Rasberry Pi. I have a concern.

Central and peripheral disconnect after exchange LTK and establish pairing.
Then re-connect them, they will exchange TK and STK again.
Can I skip encription procedure for TK and STK?

sandeepmistry, 2016-02-21:
@BYAKheee there is no persistent key mechanism in noble at this time.

TK and STK will only be exchanged if a "secure" characteristic is read. Can you please confirm your peripheral has a secure characteristic and you are trying to access it?

BYAKheee, 2016-02-29:
Thanks response!

OK, I got stable connection by another way.
I think bleno/noble's SMP conflict with kernel SMP smoetimes.

closed 2016-02-29 by BYAKheee