noble/noble #336: Emulating gatttool usage of handles in noble

LarsBuur, 2016-02-19:
I have an proprietary piece of hardware (device) that uses BT4LE for communication.

The device streams data from multiple sensors (gyro, accelerators) etc. using a notify subscription on a specific handle after it has received a command on a separate handle. The device has multiple commands that changes which data are streamed.

Our hardware experts are able to start the stream of data using gatttool - but I am unable to achieve the same using noble. 

I have tried to reproduce the steps here:

The first thing that happens is that the device are asked to list the characteristics and the accompanying handles using this command:
`gatttool -i hci0 -t random -b EE:CA:F9:D5:4A:F1 --characteristics`

This yields this response:
```
handle = 0x0002, char properties = 0x0a, char value handle = 0x0003, uuid = 00002a00-0000-1000-8000-00805f9b34fb
handle = 0x0004, char properties = 0x02, char value handle = 0x0005, uuid = 00002a01-0000-1000-8000-00805f9b34fb
handle = 0x0006, char properties = 0x02, char value handle = 0x0007, uuid = 00002a04-0000-1000-8000-00805f9b34fb
handle = 0x000a, char properties = 0x12, char value handle = 0x000b, uuid = 00002a19-0000-1000-8000-00805f9b34fb
handle = 0x000e, char properties = 0x12, char value handle = 0x000f, uuid = e4f00031-5829-11e4-8e67-0002a5d5c51b
handle = 0x0013, char properties = 0x12, char value handle = 0x0014, uuid = 674e1524-9344-11e5-b303-0002a5d5c51b
handle = 0x0017, char properties = 0x0a, char value handle = 0x0018, uuid = 674e1525-9344-11e5-b303-0002a5d5c51b
handle = 0x001a, char properties = 0x0a, char value handle = 0x001b, uuid = 674e1526-9344-11e5-b303-0002a5d5c51b
handle = 0x001d, char properties = 0x0a, char value handle = 0x001e, uuid = 674e1527-9344-11e5-b303-0002a5d5c51b
```

My hardware technician tells me that value handle "0x0014" is the data subscription handle and that the "0x001e" is the command handle. A strange piece of information is that the data subscription notify handle is really the data subscription handle +2
This means that we should able to subscribe to notifications on the data channel (0x0014) - but really listening to (0x0016).  This should just be common knowledge - but it was not to me :-)

This command enables listening for notify messages on the handle 0x16 sending the value command 0100.
`/usr/bin/gatttool --adapter hci0 -t random -b EE:CA:F9:D5:4A:F1 --char-write-req -a 0x16 -n 0100`

Sending the value command 0000 stops the notification,

Afterwards the command handle can be told to start streaming using this command. 
`/usr/bin/gatttool --adapter hci0 -t random -b EE:CA:F9:D5:4A:F1 --char-write-req -a 0x1e -n 03 --listen`

I have written this small node program to try to achieve the same result without any luck.
```
var noble = require("noble");
var _ = require("underscore");

// UUID for characteristics
var DATA_UUID = "674e1524934411e5b3030002a5d5c51b";
var SETTIME_UUID = "674e1526934411e5b3030002a5d5c51b";
var COMMAND_UUID = "674e1527934411e5b3030002a5d5c51b";

var dataCharacteristic = null;
var setTimeCharacteristic = null;
var commandCharacteristic = null;

noble.on("stateChange", function (state) {
    if (state === "poweredOn") {
        noble.startScanning();
    } else {
        noble.stopScanning();
    }
});

var validDevices = ['ee:ca:f9:d5:4a:f1'];
var RSSI_THRESHOLD = -90;

noble.on("discover", function (peripheral) {
    if (_.indexOf(validDevices, peripheral.address) >= 0 && peripheral.rssi > RSSI_THRESHOLD) {
        noble.stopScanning(); // found one. Lets explore.
        console.log(
            " devkce discovered (" + peripheral.id +
            " with address <" + peripheral.address + ">, " + peripheral.addressType + ">," +
            " connectable " + peripheral.connectable + "," +
            " RSSI:" + peripheral.rssi +
            " name:" + peripheral.advertisement.localName + "\n");
    }
    noble.on('warning', function (message) {
        console.log("warning : " + message);
    });
    connectPeripheral(peripheral);
});

var connectPeripheral = function (peripheral) {
    peripheral.on('disconnect', function () {
        console.log("disconneting");
        process.exit(0);
    });

    peripheral.connect(function (error) {
        if (error) {
            console.log("error connecting:" + error);
        } else {
            peripheral.discoverServices([], function (error, services) {
                _.each(services, function (service) {
                    characteristicsDiscovery(service, peripheral);
                });
            });
        }
    });
};

var characteristicsDiscovery = function (service, peripheral) {
    service.discoverCharacteristics([], function (error, characteristics) {
        var characteristicIndex = 0;
        _.each(characteristics, function (c) {
            switch (c.uuid) {
                case DATA_UUID :
                    dataCharacteristic = c;
                    break;
                case SETTIME_UUID:
                    setTimeCharacteristic = c;
                    break;
                case COMMAND_UUID:
                    commandCharacteristic = c;
                    break;
                default :
                // ignored for now
                //console.log(_.pick(c, ["uuid", "name", "type", "properties", "descriptors"]));
            }
        });

        if (dataCharacteristic && setTimeCharacteristic && commandCharacteristic) {
            // we have a handle on things
            dataCharacteristic.on('notify', function (state) {
                console.log("dataCharacteristic notify is :" + (state ? "on" : "off"))
            });

            dataCharacteristic.on('read', function (data, isNotification) {
                console.log("data:" + data.toString('hex'));
            });

            // true lets enable notify on dataCharacteristic
            dataCharacteristic.notify(true, function (error) {
                if (error) {
                    console.log("notify error: " + error)
                }
                console.log('data channel notification on');
                setTimeout(function () {

                    commandCharacteristic.write(new Buffer(0x03), true, function (e) {
                        if (e) {
                            console.log(e)
                        }
                    });

                    // this is an attempt to write to the handle directly - but no luck....
                    peripheral.writeHandle(0x1e, new Buffer(0x03), true, function (e) {
                        if (e) {
                            console.log(e);
                        }
                    });

                    console.log("command set. - now data should be streaming...")
                }, 900);
            });
        }
    });
    setTimeout(function () {
        peripheral.disconnect();
    }, 10000);
};
```

this is the output from the console.

```
node commands.js
 devkce discovered (a00a7f3005614adf811eacbba2c3756b with address <ee:ca:f9:d5:4a:f1>, unknown>, connectable true, RSSI:-65 name:BC10EECAF9D54AF1

dataCharacteristic notify is :on
data channel notification on
command set. - now data should be streaming...
disconneting
```

I have also tried without the setTimeout - and multiple timeout values.

Both of the writes to the device using gatttool are using "char-write-req" which suggest that it is a request. Therefor I have tried both 
`commandCharacteristic.write(new Buffer(0x03), true, function (e)`
and
`commandCharacteristic.write(new Buffer(0x03), false, function (e)`

on both the write characteristic and writeHandle... 

Any suggestions would be much appreciated. I have tried all the different settings, handle writes that I can think of. I might just be some silly error / misunderstanding that I have made. 

I am using a MacBook Pro (Retina, 13-inch, Late 2013) (2,8 GHz Intel Core i7) on OSX El Capitan (10.11). I have tried the same code on ubuntu 14.4 and raspberry pi (debian) with same result.

sandeepmistry, 2016-02-21:
@LarsBuur
```
commandCharacteristic.write(new Buffer(0x03), false, function (e)
```

is close, but please look at the Node.js Buffer documentation: https://nodejs.org/api/buffer.html#buffer_class_buffer

You are creating a buffer of size 3, which will not be initialized with any value in particular. I believe you want to use: ```new Buffer([0x03])``` instead to create a buffer 1 byte in length with a value of 3.

I also recommend staying away from the write handle API's.

closed 2016-02-21 by sandeepmistry

LarsBuur, 2016-02-21:
Thank you very much @sandeepmistry - that was indeed the problem.

sandeepmistry, 2016-02-21:
@LarsBuur great! Thanks for following up.

craigmw, 2016-10-26:
I have a question related to this issue. If I issue the following command to a BLE sensor, it returns the battery value on that sensor:

gatttool -b 88:c2:55:ac:26:cc --char-read -a 0x48

Which returns:

Characteristic value/descriptor: 3f

So, the characteristic's handle is apparently 0x48. Do I need to determine the characteristic's UUID or can I simply provide the handle 0x48 in noble to read this value?

Thanks!

sandeepmistry, 2016-10-30:
@craigmw 

>  Do I need to determine the characteristic's UUID or can I simply provide the handle 0x48 in noble to read this value?

Determine the characteristics UUID and the service it belongs to is the way to go.