noble/noble #337: Change Char.notify to subscribe

VinceGa, 2016-02-19:
In Bleno,there are onSubscribe: null, // optional notify/indicate subscribe handler, function(maxValueSize, updateValueCallback) { ...}
    onUnsubscribe: null, // optional notify/indicate unsubscribe handler, function() { ...}
    onNotify: null // optional notify sent handler, function() { ...}
    onIndicate: null // optional indicate confirmation received handler, function() { ...}  

But in Noble, we only have characteristic.notify(notify[, callback(error)]); // notify is true|false

I suggest we change the method to characteristic.subscribe(subscribe[, callback]).  It will reflect what we do in a better way.

..and before that, we need a simple way to check if the char is subscriptable. (check for CCCD)

even better, we can separate Notification and Indication in Noble.

Thanks.

sandeepmistry, 2016-02-21:
@VinceGa 

> But in Noble, we only have characteristic.notify(notify[, callback(error)]); // notify is true|false
>
> I suggest we change the method to characteristic.subscribe(subscribe[, callback]). It will reflect what we do in a better way.

This is a good point however I would prefer the following:

 * ```characteristic.subscribe(callback(error)```` which calls ```characteristic.notify(true, callback``` internally.
 * ```characteristic.unsubscribe(callback(error)```` which calls ```characteristic.notify(false, callback``` internally.

> ..and before that, we need a simple way to check if the char is subscriptable. (check for CCCD)

Can't ```characteristic.properties``` be used for this? https://github.com/sandeepmistry/noble#characteristics-discovered

> even better, we can separate Notification and Indication in Noble

We're stuck with them combined, as the OS X CoreBluetooth API does not separate them ... 

https://developer.apple.com/library/mac/documentation/CoreBluetooth/Reference/CBPeripheral_Class/index.html#//apple_ref/occ/instm/CBPeripheral/setNotifyValue:forCharacteristic:

VinceGa, 2016-02-22:
Yep, agreed on the new signature.
Yes, the Char Discovered event could be used for this.

Thanks a lot!

sandeepmistry, 2016-02-27:
@VinceGa great, would you be able to submit a PR for the new API's?

sandeepmistry, 2016-05-24:
Added in https://github.com/sandeepmistry/noble/commit/472c6036eaa4e9c236e91e23987289d1c074731f

closed 2016-05-24 by sandeepmistry