noble/noble #339: About publishing project

hedywings, 2016-02-24:
Since the requirement of my project, I need some functions like change scan parameters, change connect parameters and connect cancel..., but I can't test them on OS X, so I think you may not accept my pull request. Therefore I fork your noble project to my github and add some functions that I need in my project. May I publish this modified project to npm?

Thanks.

sandeepmistry, 2016-02-27:
@hedywings thank you for contacting me about this. Do you absolutely need to publish your module to npm?

If you do not plan to maintain your fork, I would prefer it was not published to npm. For example, [noble-nan](https://www.npmjs.com/package/noble-nan) a fork of noble was published about a year ago and has not received any updates, it also shows up when you search for noble on npm.

A few options:

1) Refer to your fork on Github when using npm/package.json: ```npm install --save hedywings/noble```.

2) Use a scoped package name on npm: http://blog.npmjs.org/post/116936804365/solving-npms-hard-problem-naming-packages - I believe you will have to modify the name in package.json to be ```@hedywings/noble```.

Let me know what you think. Again thanks for brining this up with me.

hedywings, 2016-03-06:
Thank you for provide options for me, I don't even know these methods!!

After consideration I have decided to adopt the first approach to refer my fork!

Thanks a lot~

closed 2016-03-06 by hedywings