noble/noble #340: Missing services and characteristics on OS X El Capitan

daetherius, 2016-02-25:
Mac OS X El Capitan
MacbookPro mid 2012
BLE Adaptor: Broadcom 20702A3

I'v been trying to connect to an Ollie by Sphero, but I'm not getting any data for the services and characteristics when running the `peripheral-explorer.js` tool. This is the output:
```
peripheral with ID 6d700ca9b9b944348adeb122c19600ca found
  Local Name        = 2B-7683
  TX Power Level    = -10
  Service Data      =
  Service UUIDs     =

services and characteristics:
```
However, I'm able to get this information when using the Bluetooth explorer from the Hardware IO Tools, as shown here http://i.imgur.com/shdtYCn.png

Running  `sudo DEBUG=* node ./node_modules/noble/examples/peripheral-explorer.js 6d700ca9b9b944348adeb122c19600ca` outputs the following:

https://gist.github.com/daetherius/6df79ca6c28964740893

Any ideas? Thanks in advance!

sandeepmistry, 2016-02-27:
@daetherius what happens when if you try the LightBlue app from the Mac App store?

CoreBluetooth or blued might be confused, so you can try:

1) Killing and restarting blued
2) Clearing the CoreBluetooth cache: http://stackoverflow.com/a/23573314/2020087

daetherius, 2016-02-29:
@sandeepmistry Thanks for your response. Actually, I've already tried both solutions you provided before submitting this issue.

LightBlue app seems to retrieve a list of services correctly http://i.imgur.com/TpGzS2M.png

daetherius, 2016-03-04:
Any news? I'm still having this problem.

b44p, 2016-03-07:
Hey - having same issues as referenced in above from @deadprogram

deadprogram, 2016-03-07:
Actually, @daetherius is the one with that problem. Interestingly, I am able to connect from my own recent model Macbook Air.

b44p, 2016-03-11:
Still having the issue. Oddly enough my MAC does show that it is connected to what looks like the Ollie but then the script just hangs saying `2016-03-11T13:58:03.171Z : [Robot 1] - Starting device 'ollie'.`

http://i.imgur.com/GWNigCS.png

sandeepmistry, 2016-04-24:
@daetherius @b44p 

```
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 40,
  "kCBMsgArgs": {
    "kCBMsgArgDeviceUUID": {
      "type": "Buffer",
      "data": [
        109,
        112,
        12,
        169,
        185,
        185,
        68,
        52,
        138,
        222,
        177,
        34,
        193,
        150,
        0,
        202
      ]
    },
    "kCBMsgArgResult": 23
  }
} +151ms
```

This message is blued sending a disconnect event. I would suggest looking the Xcode Hardware IO tools PacketLogger app for any hints. Unfortunately this is out of scope for noble. It could also be caused by the firmware version running on the peripheral.

I'm going to close this for now, unless you have more information to narrow down it to something specifically in noble and not OS X/blued.

closed 2016-04-24 by sandeepmistry