noble/noble #345: Bug: Noble discovers duplicate services after premature disconnect

ayhoung, 2016-03-03:
```
peripheral.connect(function(error) {
        log('Discovering services for peripheral ' + peripheral.advertisement.localName);
        peripheral.discoverServices(serviceUUIDs, function(error, services) {
            var loopdService = services[0];
            log('Discovering characteristics for peripheral ' + peripheral.advertisement.localName);
            loopdService.discoverCharacteristics(characteristicsUUID, function(error, characteristics) {
                loopdCharacteristic = characteristics[0];
                loopdCharacteristic.notify(true, function(error) {
                    //issue command
                });
            });
        });
    });
```

This is what I get for an output:
`[2016-03-03T09:04:19.093Z]  INFO: badge-state-viewer/2414 on edison126: Discovering services for peripheral 6BDF3ABB (env=production)
[2016-03-03T09:04:19.196Z]  INFO: badge-state-viewer/2414 on edison126: ======================== (env=production)
[2016-03-03T09:04:19.197Z]  INFO: badge-state-viewer/2414 on edison126: peripheral disconnected! (env=production)
[2016-03-03T09:04:19.198Z]  INFO: badge-state-viewer/2414 on edison126: ======================== (env=production)
[2016-03-03T09:04:27.162Z]  INFO: badge-state-viewer/2414 on edison126: command for badge 6BDF3ABB called with <Buffer f7 01 04> (env=production)
[2016-03-03T09:04:27.996Z]  INFO: badge-state-viewer/2414 on edison126: inside connectBadge function (env=production)
[2016-03-03T09:04:28.984Z]  INFO: badge-state-viewer/2414 on edison126: Discovering services for peripheral 6BDF3ABB (env=production)
[2016-03-03T09:04:29.189Z]  INFO: badge-state-viewer/2414 on edison126: Discovering characteristics for peripheral 6BDF3ABB (env=production)
[2016-03-03T09:04:29.191Z]  INFO: badge-state-viewer/2414 on edison126: Discovering characteristics for peripheral 6BDF3ABB (env=production)
[2016-03-03T09:04:29.192Z]  INFO: badge-state-viewer/2414 on edison126: Discovering characteristics for peripheral 6BDF3ABB (env=production)`

ayhoung, 2016-03-03:
here's an hcidump:
`
2016-03-03 09:23:31.387945 > HCI Event: LE Meta Event (0x3e) plen 35
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 7A:8B:61:6C:F2:2F (Random)
      Flags: 0x06
      Unknown type 0xff with 18 bytes data
      RSSI: -75
2016-03-03 09:23:31.388424 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr 7A:8B:61:6C:F2:2F (Random)
      RSSI: -75
2016-03-03 09:23:31.418321 > HCI Event: LE Meta Event (0x3e) plen 35
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 7A:8B:61:6C:F2:2F (Random)
      Flags: 0x06
      Unknown type 0xff with 18 bytes data
      RSSI: -75
2016-03-03 09:23:31.418541 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr 7A:8B:61:6C:F2:2F (Random)
      RSSI: -75
2016-03-03 09:23:31.424389 > HCI Event: LE Meta Event (0x3e) plen 35
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 4C:7C:7C:20:98:E0 (Random)
      Flags: 0x06
      Unknown type 0xff with 18 bytes data
      RSSI: -90
2016-03-03 09:23:31.424722 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr 4C:7C:7C:20:98:E0 (Random)
      RSSI: -93
2016-03-03 09:23:31.451777 > HCI Event: LE Meta Event (0x3e) plen 35
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 7A:8B:61:6C:F2:2F (Random)
      Flags: 0x06
      Unknown type 0xff with 18 bytes data
      RSSI: -74
2016-03-03 09:23:31.452255 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr 7A:8B:61:6C:F2:2F (Random)
      RSSI: -75
2016-03-03 09:23:31.481921 > HCI Event: LE Meta Event (0x3e) plen 35
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 7A:8B:61:6C:F2:2F (Random)
      Flags: 0x06
      Unknown type 0xff with 18 bytes data
      RSSI: -75
2016-03-03 09:23:31.499557 > HCI Event: LE Meta Event (0x3e) plen 27
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr B8:78:2E:28:66:AB (Public)
      Flags: 0x1a
      Unknown type 0xff with 10 bytes data
      RSSI: -70
2016-03-03 09:23:31.499866 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr B8:78:2E:28:66:AB (Public)
      RSSI: -70
2016-03-03 09:23:31.514444 > HCI Event: LE Meta Event (0x3e) plen 35
[... 966 lines cut ...]
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 4C:7C:7C:20:98:E0 (Random)
      Flags: 0x06
      Unknown type 0xff with 18 bytes data
      RSSI: -91
2016-03-03 09:23:34.543270 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr 4C:7C:7C:20:98:E0 (Random)
      RSSI: -92
2016-03-03 09:23:34.558421 > HCI Event: LE Meta Event (0x3e) plen 35
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 66:ED:BE:9F:80:51 (Random)
      Flags: 0x06
      Unknown type 0xff with 18 bytes data
      RSSI: -80
2016-03-03 09:23:34.558642 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr 66:ED:BE:9F:80:51 (Random)
      RSSI: -77
^C
root@edison126:~#
`

sandeepmistry, 2016-03-06:
This is related to https://github.com/sandeepmistry/noble/issues/193

rzr, 2021-03-02:
Please can you update status of this issue on this latest release: 

https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-12