noble/noble #346: " Cannot find module 'xpc-connection' from '.node_modules\noble\lib' " when creating win executable with nexe and enclosejs.

aditya02maurya, 2016-03-03:
Hi,
     wrote a simple script using NOBLE. Its works when run with "node scriptname.js" from cmd line but **fails while creating windows exe from the script.
I used NEXE and ENCLOSE but both failed due to noble!**
(using WINDOWS 8.1, NODE v4.1.2 (also v5.6), NPM v3.7.5)

Please find the command and error below, why does noble need Mac dependency on windows?
  
F:\Adi\A Web Learning\web Bisoft>nexe -i check1.js -o bisExe1 -t ./tmp/nexe
----> framework => nodejs
----> bundle F:\Adi\A Web Learning\web Bisoft\check1.js
----> found browserify in C:\Users\Adi\AppData\Roaming\npm\node_modules\nexe\node_modules\browserify
\bin\cmd.js
**Error: Cannot find module 'xpc-connection' from 'F:\Adi\A Web Learning\web Bisoft\node_modules\noble
\lib\mac'**
    at C:\Users\Adi\AppData\Roaming\npm\node_modules\nexe\node_modules\resolve\lib\async.js:46:17
    at process (C:\Users\Adi\AppData\Roaming\npm\node_modules\nexe\node_modules\resolve\lib\async.js
:173:43)
    at ondir (C:\Users\Adi\AppData\Roaming\npm\node_modules\nexe\node_modules\resolve\lib\async.js:1
88:17)
    at load (C:\Users\Adi\AppData\Roaming\npm\node_modules\nexe\node_modules\resolve\lib\async.js:69
:43)
    at onex (C:\Users\Adi\AppData\Roaming\npm\node_modules\nexe\node_modules\resolve\lib\async.js:92
:31)
    at C:\Users\Adi\AppData\Roaming\npm\node_modules\nexe\node_modules\resolve\lib\async.js:22:47
    at FSReqWrap.oncomplete (fs.js:82:15)

....> ERROR: Browserify failed to launch

sandeepmistry, 2016-03-06:
@aditya02maurya please see the nexe read me: https://github.com/jaredallard/nexe#doesnt-support-dynamic-require-statements

noble uses "dynamic" require statements, nexe read me says:

> Doesn't support dynamic require statements

Marking this as won't fix. You'll also run into issues with native modules as per: https://github.com/jaredallard/nexe#doesnt-support-native-modules

closed 2016-03-06 by sandeepmistry

aditya02maurya, 2016-03-09:
Thanks for your reply.