noble/noble #347: Characteristic notifications stop randomly

billskeen68, 2016-03-03:
Using latest Broadcom drivers from Zadig and libraries for Node, Noble, bluez, libusb, etc.. 
Running on both Windows 10 and Debian Jessie.

On both systems, after a random amount of time the notifications stop. On windows, I receive about 45-60 events before it dies and on Linux around 5-6. I have similar setup running on rPI and python using BLED112 without any issues, so I am fairly certain the BLE devices are not stopping.

What have I forgotten to do? I know, tough question given I did not listed every step taken but any help appreciated.

Bill

billskeen68, 2016-03-04:
My device is a Mio Alpha 2 Heartrate wristband. It appears to be having an issue similar to issue #296. I timed it from peripheral.connect() to the last heartrate notification and it is ~60 seconds. This is on Windows 10.

sandeepmistry, 2016-03-06:
@billskeen68 on Windows certainly does sound like #296.

Let's tackle Linux first, please provide an HCI dump capture of the entire sequence: ```sudo hcidump -t -x```

PimButton, 2016-03-13:
As the person who created #296: in my case the notifications did not just stop, the connection was actually disconnected. This was caused by my peripheral who sends a Connection parameter update request and disconnects if the master does not reply within 60 seconds. Since Noble does not allow you to change the Connection Parameters this request is ignored.

sandeepmistry, 2016-03-20:
I'm closing this due to lack of activity, if you are still interested please provide the requested info from https://github.com/sandeepmistry/noble/issues/347#issuecomment-192966723.

closed 2016-03-20 by sandeepmistry