noble/noble #348: Unable to scan for peripherals on C.H.I.P.

deadprogram, 2016-03-07:
I installed Noble on C.H.I.P. using nvm and Node 4.3.2

I was able to run the `examplesadvertisement-discovery.js` program without any apparent error, however it never returns any output.

Here is the `DEBUG=*` log I captured. https://gist.github.com/deadprogram/2af40982c1212912d613

The HCI dump is pretty large (over 2MB). Let me know if you want me to put it somewhere.

Thanks!

deadprogram, 2016-04-11:
Hi @sandeepmistry sorry to bother you. I know you've played with C.H.I.P. on the GPIO side of things. Did you happen to run Noble on it, and if so, what if anything did you need to do to unblock it? Thanks!

sandeepmistry, 2016-04-24:
@deadprogram have you compared the HCI dump from the C.H.I.P. to a working Linux system?

deadprogram, 2016-04-25:
Hi, @sandeepmistry I'll setup my C.H.I.P. again for some experiments this week, and get back to you on that.

sandeepmistry, 2016-06-06:
@deadprogram ping ...

sandeepmistry, 2016-09-26:
Closing this for now due to lack of activity.

closed 2016-09-26 by sandeepmistry