noble/noble #350: Duplicate messages received from bluetooth device

mdlampi, 2016-03-08:
On Mac OS X 10.10.5 Yosemite we have observed that NOBLE sometimes returns duplicate or even triplicate messages.

The Mac Packet Logger program shows there should be only one message according to the timestamps of the data it logs as well as the timestamps logged by the NOBLE application.

Other than placing a higher level filter comparing previous input to the latest input, is there a fix for this?

sandeepmistry, 2016-03-13:
@mdlampi please provide sample code and output.

mdlampi, 2016-03-14:
@sandeepmistry Here is a log of the program output. Attached are the bluetooth routines that perform the reads via NOBLE, and Mac Packet Logger logs taken during the time the program was executed. There are 3 "Got duplicate" occurrences logged, along with the hex strings for the messages returned by NOBLE.

Let me know if you need any more information.

Thanks!

[Archive.zip](https://github.com/sandeepmistry/noble/files/172436/Archive.zip)

sandeepmistry, 2016-03-20:
@mdlampi I don't think the zip file contains all the code.

Anyways, it would really speed things up if you provided a minimal way to reproduce the issue in pure javascript (no coffeescript).

mdlampi, 2016-03-31:
@sandeepmistry We had a new team take a look at the bluetooth stack on the device. There were some situations where packets could be overwritten and this is what probably caused the duplicate messages.

Thank you anyway!

closed 2016-03-31 by mdlampi