noble/noble #351: Error: Cannot find module 'bluetooth-hci-socket'

RandoSY, 2016-03-09:
Howdy!

I followed the instructions carefully from here... (on Windows 10)

    https://www.youtube.com/watch?v=mL9B8wuEdms

Quite  a few warnings/error? during install, but ignored as instructed.

When I attempt to discovery my Vivofit via BLE I get this error:

C:\Users\Randall\Desktop\Arduino_Experimenter_Guide_for_NodeJS\Toolchain\noble-master\examples>node advertisement-discovery.js
module.js:327
    throw err;
    ^
**Error: Cannot find module 'bluetooth-hci-socket'**
    at Function.Module._resolveFilename (module.js:325:15)
    at Function.Module._load (module.js:276:25)
    at Module.require (module.js:353:17)
    at require (internal/module.js:12:17)
    at Object.<anonymous> (C:\Users\Randall\Desktop\Arduino_Experimenter_Guide_f
or_NodeJS\Toolchain\noble-master\lib\hci-socket\hci.js:6:26)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Module.require (module.js:353:17)

Any suggestions, much appreciated.

R.

RandoSY, 2016-03-09:
Logically enough I did try and install npm -install "bluetooth-hci-socket" This fails, most probably with ...

gyp ERR! node -v v4.3.2
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok
npm WARN optional dep failed, continuing bluetooth-hci-socket@0.4.2

I attempted to upgrade node-gyp globally via this set of instructions, very carefully...

    https://github.com/nodejs/node-gyp/wiki/Updating-npm's-bundled-node-gyp

But still no joy, :( ...

R.

closed 2016-03-09 by RandoSY

sandeepmistry, 2016-03-13:
@RandoSY please provide the ```npm-debug.log``` and/or full output of the npm install if you'd like further assistance.