noble/noble #353: stateChange not called on Windows 8 and 10

PimButton, 2016-03-12:
## Background
I'm using Noble in combination with Electron so that my compiled program can be easily installed and run. I develop on windows 7 and everything works great on that environment. I'm using the Asus BT-400 adapter with the correct driver installed by the Zadig tool.

## Problem
On various Windows 8 and 10 based computers however, in about 3 out of 5 times the program is started it seems like the stateChange function is not being called, and therefore a scan does not start. 

## Analysis
I have run a few tests, and Noble does detect the BT-400 adapter because as soon as I pull it out of the computer and start the program, Noble will error and say "No compatible Bluetooth 4.0 adapter found". This always happens when the adapter is not in the computer. When the adapter ís in the computer, it will work 2 out of 5 times without error, and the other 3 out of 5 times it will just sit there and wait. I assume it waits for the stateChange function, cause there is no error and nothing in the log file. When I print ```noble.state``` when the scan does not start, the state is ```unknown```. 

## Solution?
So far I have not been able to find a solution. Just starting the program a bunch of times works in a way that it will eventually start scanning but this makes the program very impracticable. @sandeepmistry do you have a Windows computer to test on? I could send you the compiled program if you need to test with that.

sandeepmistry, 2016-03-20:
@PimButton please provide debug logs:

```
set DEBUG=*
node <file>.js
```

It would also be worthwhile to try the master version of node-bluetooth-hci-socket, I'm wondering if https://github.com/sandeepmistry/node-bluetooth-hci-socket/commit/39af043593335fd985c01d9e9426c25d5447078e might fix this.

PimButton, 2016-03-20:
Well the issue seems fixed. I read trough https://github.com/sandeepmistry/node-bluetooth-hci-socket/issues/7 and noticed that the electron version I was using (v0.30.4) had no support for NAN2 yet. I recompiled everything for Electron v0.37.2 and it now works as expected. Very strange that this had only impacted Win 7 and Win 10 but I'm glad it is solved :)

closed 2016-03-20 by PimButton