noble/noble #354: Bug: Value is out of bounds when trying to connect to same peripheral twice

ayhoung, 2016-03-13:
Noticed an interesting bug where if you try to connect to the same peripheral twice, noble generates an error in the buffer saying "Command Disallowed", and enters a state of non-action. HCI doesn't generate any other events after entering this state.

ayhoung, 2016-03-13:
This also seems to happen when trying to connect to a non-existent peripheral.

Here's the stacktrace:
TypeError: value is out of bounds
      at TypeError (<anonymous>)
      at checkInt (buffer.js:705:11)
      at Buffer.writeUInt16LE (buffer.js:722:5)
      at Hci.disconnect (/home/root/badge-state-viewer/noble/lib/hci-socket/hci.js:290:7)
      at NobleBindings.disconnect (/home/root/badge-state-viewer/noble/lib/hci-socket/bindings.js:57:13)
      at Noble.disconnect (/home/root/badge-state-viewer/noble/lib/noble.js:164:18)
      at Peripheral.disconnect (/home/root/badge-state-viewer/noble/lib/peripheral.js:57:15)
      at disconnectPeripheral (/home/root/badge-state-viewer/lib/sync-manager.js:417:20)
      at null._onTimeout (/home/root/badge-state-viewer/lib/sync-manager.js:352:13)
      at Timer.listOnTimeout [as ontimeout] (timers.js:112:15)

ayhoung, 2016-03-13:
also seeing this in the hcidump:
2016-03-13 09:11:25.260164 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 20:73:6A:10:05:2E type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6
2016-03-13 09:11:25.261513 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x0c ncmd 1
    Error: Command Disallowed

ayhoung, 2016-03-13:
Actually realized it was my problem! I was trying to disconnect with a peripheral when there was no connection, resulting in an error. I have another question though, which is how to cancel a connection attempt by peripheral.connect()?

closed 2016-03-13 by ayhoung

sandeepmistry, 2016-03-13:
https://github.com/sandeepmistry/noble/issues/229 is currently tracking the ability to cancel a connection, it is not supported yet.