noble/noble #355: Cannot connect to peripherals after a while

davidmessem, 2016-03-15:
Connecting to peripherals seems to hang in the 'connecting' or 'disconnecting' state after using noble to scan and connect to devices for a while. I can connect to the peripherals using other centrals at the same time as the as the connections are hanging.

I am using a RaspberryPi with a Cambridge Silicon Radio Bluetooth 4 USB dongle, and a RaspberryPi 3 with built in bluetooth. I use noble to scan for devices, connect to them and discover services, characteristics and read a characteristic, and finally disconnect from the peripheral.

I have attached an hci dump
[hci_dump.txt](https://github.com/sandeepmistry/noble/files/173407/hci_dump.txt)

sandeepmistry, 2016-03-20:
@davidmessem please provide a minimal code example to reproduce the issue and full HCI dump capture.

sandeepmistry, 2016-04-03:
I'm closing this due to lack of feedback, please re-open with the requested info. if you are still interested.

closed 2016-04-03 by sandeepmistry