noble/noble #357: Maximum simultaneous connections

pdaddy26, 2016-03-15:
I was wondering where does this limit come from? Isn't it suppose to be a limit from the hardware?

sandeepmistry, 2016-03-20:
@pdaddy26 yes this limit is both dependent on the hardware and firmware running on the adapter.

closed 2016-03-20 by sandeepmistry

Kazjon, 2016-08-10:
We're trying to use an adapter that is supposed to have a much higher simultaneous connection limit (over 50), but still running into the 5-device limit in noble.  We're on raspbian on a Raspberry Pi 3 using an external bluetooth dongle.  Is it possible that there's something in noble which is limiting us artificially?  Is there any way we can print out a more verbose log of why connections are failing?  Thanks!

sandeepmistry, 2016-08-14:
@Kazjon nope, I believe it's all hardware and firmware dependent. HCI dump will provide the info. you need: ```sudo hcidump -t -x```