noble/noble #358: Unable to connect to multiple devices

pdaddy26, 2016-03-18:
Hi, 

I have 5 devices and most of the time it will only connect to 1 or 2 devices. Sometimes 3, but rarely more. I'm running ubuntu 14.04 TLS. 

Here is my code if it can help:
```
var noble = require('noble');
var os = require('os');

noble.on('stateChange', function(state) {
    console.log(state);
    if (state === 'poweredOn') {
        noble.startScanning();
    } else {
    }
});

noble.on('scanStart', function() {
    console.log('scanStart');
});

noble.on('scanStop', function() {
    console.log('scanStop');
});

noble.on('discover', function(peripheral) {

    var localName = peripheral.advertisement.localName;
    console.log("discover ", localName);

    if(localName !== undefined && localName !== "" && localName.substring(0, "MY".length) === "MY") {
        console.log("Connecting ", localName);

        peripheral.on('connect', function() {
            console.log(localName, 'connected');
        });

        peripheral.on('disconnect', function() {
            console.log(localName, 'disconnected');
        });

        peripheral.connect(function(error) {
            if(error) {
                console.log(error);
                return;
            }
            console.log(localName, 'connected!'); 
	    return;
        });
    }

});
```
From what I've seen it will not discover the others. As if as soon as it connects to 1 device, it is not discovering the others thus not making an attempt to connect. 

Also, would it work with bluez 5.xx?

sandeepmistry, 2016-03-20:
@pdaddy26 try adding a ```noble.startScanning();``` inside the ```peripheral.connect(function(error) {```` callback.

This might be related to https://github.com/sandeepmistry/node-sensortag/issues/37.

pdaddy26, 2016-03-21:
Thanks that worked. I noticed the same issue with hcitool lescan, so I'm guessing this might be an issue with bluez?

Anyhow, thanks for the help/

closed 2016-03-21 by pdaddy26

sandeepmistry, 2016-03-28:
@pdaddy26 I think it's a firmware issue, some Bluetooth adapters won't continue to scan after a connection has been established.

pdaddy26, 2016-03-29:
Do you think you can emit a scanStop when this happens?

sandeepmistry, 2016-04-03:
@pdaddy26 maybe, we'll have to see what HCI data the adapter sends.

Could you please capture a HCI dump (```sudo hcidump -t -x```) while:
 * Running: ```sudo hciconfig hci0 reset```
 * Running your Node.js app

Unfortunately now of my HCI adapters have this issue, so I cannot reproduce locally. I'm wondering if the adapter reports if is can continue to scan after connecting via some feature or version command response.

pdaddy26, 2016-04-06:
Here is the dump when running my node js app. 

```
HCI sniffer - Bluetooth packet analyzer ver 2.5
device: hci0 snap_len: 1500 filter: 0xffffffffffffffff
2016-04-06 12:42:14.476064 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2016-04-06 12:42:14.477147 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-04-06 12:42:14.477184 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2016-04-06 12:42:14.478142 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-04-06 12:42:14.478173 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2016-04-06 12:42:14.480142 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x1000
    LMP Version: 4.0 (0x6) LMP Subversion: 0x220e
    Manufacturer: Broadcom Corporation (15)
2016-04-06 12:42:14.480170 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00 
2016-04-06 12:42:14.481141 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2016-04-06 12:42:14.481174 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2016-04-06 12:42:14.483140 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2016-04-06 12:42:14.483173 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2016-04-06 12:42:14.485138 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 5C:F3:70:62:4B:A3
2016-04-06 12:42:14.485166 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-04-06 12:42:14.486139 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-04-06 12:42:14.486171 < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
2016-04-06 12:42:14.487140 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2016-04-06 12:42:14.502562 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
2016-04-06 12:42:14.505132 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-04-06 12:42:14.550137 > HCI Event: LE Meta Event (0x3e) plen 35
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 00:A0:50:07:23:2F (Public)
      Flags: 0x06
      Complete local name: 'HX-00042182'
      TX power level: 0
      Shortened service classes: 0x180d
      RSSI: -85
2016-04-06 12:42:14.554126 > HCI Event: LE Meta Event (0x3e) plen 29
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr 00:A0:50:07:23:2F (Public)
      Complete local name: 'HX-00042182'
      Shortened service classes: 0x180d
      RSSI: -85
2016-04-06 12:42:14.559992 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 00:A0:50:07:23:2F type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6
2016-04-06 12:42:14.560127 > HCI Event: LE Meta Event (0x3e) plen 33
    LE Advertising Report
      ADV_SCAN_IND - Scannable undirected advertising (2)
      bdaddr EC:24:B8:4D:30:F7 (Public)
      Flags: 0x06
      Unknown type 0x07 with 16 bytes data
      RSSI: -87
2016-04-06 12:42:14.561143 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2016-04-06 12:42:14.615142 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role master
      bdaddr 00:A0:50:07:23:2F (Public)
2016-04-06 12:42:14.615315 < HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2
  40 00 
2016-04-06 12:42:14.618132 > HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
2016-04-06 12:42:14.618473 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)
2016-04-06 12:42:14.622140 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-04-06 12:42:14.644144 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Read Remote Used Features Complete
      status 0x00 handle 64
      Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
2016-04-06 12:42:14.653640 < ACL data: handle 64 flags 0x00 dlen 7
    ATT: MTU req (0x02)
      client rx mtu 256
2016-04-06 12:42:14.674137 > ACL data: handle 64 flags 0x02 dlen 7
    ATT: MTU resp (0x03)
      server rx mtu 23
2016-04-06 12:42:14.805148 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
```

sandeepmistry, 2016-04-09:
@pdaddy26 thanks.

What's the output of ```hciconfig -a```? Here's mine, for an adapter that can connect and continue to scan at the same time.

```
$ sudo hciconfig -a
hci0:	Type: BR/EDR  Bus: USB
	BD Address: 00:02:72:3E:3C:B3  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN 
	RX bytes:2099 acl:0 sco:0 events:82 errors:0
	TX bytes:1665 acl:0 sco:0 commands:82 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'noname-0'
	Class: 0x620100
	Service Classes: Networking, Audio, Telephony
	Device Class: Computer, Uncategorized
	HCI Version: 4.0 (0x6)  Revision: 0x1000
	LMP Version: 4.0 (0x6)  Subversion: 0x220e
	Manufacturer: Broadcom Corporation (15)
```

It would be interesting to comparing: ```Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87```

pdaddy26, 2016-05-02:
```
hci0:	Type: BR/EDR  Bus: USB
	BD Address: 5C:F3:70:68:93:62  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN 
	RX bytes:664 acl:0 sco:0 events:42 errors:0
	TX bytes:964 acl:0 sco:0 commands:42 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'p-Aspire-3810TZ-0'
	Class: 0x6c0100
	Service Classes: Rendering, Capturing, Audio, Telephony
	Device Class: Computer, Uncategorized
	HCI Version: 4.0 (0x6)  Revision: 0x1000
	LMP Version: 4.0 (0x6)  Subversion: 0x220e
	Manufacturer: Broadcom Corporation (15)
```

sandeepmistry, 2016-05-08:
@pdaddy26 thanks, unfortunately it looks the same to me.