noble/noble #359: Noble shows just chars 5-8 from a service data uuid

rodrigosalinas, 2016-03-22:
Hi. Why Noble is not showing the complete uuid? This is my code:

```javascript
noble.on('discover', function(peripheral) {
    console.log('Data UUID: ' + peripheral.advertisement.serviceData[0].uuid);
    noble.stopScanning();
    setTimeout(function() {
        noble.startScanning();
    }, 3000);
});
```

This is what I get:

`Data UUID: 1523`

This is what I'm sending from my Android app:

`bbc21523-008e-fe6b-28ee-d6a44f9cb5da`

Thank you very much in advance.

sandeepmistry, 2016-03-28:
@rodrigosalinas I believe this is a limitation of the specification, service data UUID's are limited to 16-bits, so I imagine your Android app is not really sending the UUID you provided over the air.

Have you compared the behaviour of noble to other central stacks?

I'm closing this for now, until you provide an example of another central stack showing the complete uuid.

closed 2016-03-28 by sandeepmistry

jameshowe, 2017-07-08:
@sandeepmistry standardised service UUIDs are restricted to 16-bits, however, custom service UUIDs must be 128-bits, I believe the library is restricted in that sense.

sandeepmistry, 2017-07-31:
@jameshowe pull requests welcome to address this ...